jgaeddert/liquid-dsp #6: More liberal license?

JayKickliter, 2015-02-06:
At my last job we were in awe of liquid DSP's completeness and documentation, but we were not legally/contractually allowed to use GPL code. Would you be open to release it under a more liberal license? 

I also ask because I'm working on some [DSP](https://github.com/JayKickliter/Multirate.jl) and [communications](https://github.com/JayKickliter/Radio.jl) code for the Julia language, which can directly call c libraries. I'd love to link to Liquid, but the Julia community prefers the MIT license, and is actively trying to migrate away from all its GPL dependencies.

I'm sorry if this is an offensive request, and will completely understand if you prefer sticking to GPL.

jgaeddert, 2015-02-12:
Jay, sorry for the late response. I get tons of requests for a less restrictive license. Fortunately a new license is on the way. I've been deliberating for a while on which one to go with. I think the X11 license (commonly known as the "MIT license") is probably best suited for most people. I'd love your opinion on the matter.

gerasim13, 2015-02-12:
That's great news! Current license stopped me from using liquid-dsp in my new iOS app.

JayKickliter, 2015-02-12:
I'm not religious about open source licenses, and X11/MIT is perfect.

If the transition is just a matter of updating files, I'd be happy to help. I do know that some dependencies may be GPL, but that can just be spelled out [like this](https://github.com/JuliaLang/julia/blob/master/LICENSE.md).

andresv, 2015-05-19:
Hi, is there any news about migrating to MIT license?

jgaeddert, 2015-05-19:
Yes! I have the changes made. I can push them to github when I get home tonight.

jgaeddert, 2015-05-20:
Pushed. Enjoy!

closed 2015-05-20 by jgaeddert

JayKickliter, 2015-05-20:
Thanks so much!

andresv, 2015-05-20:
Thanks.

PS: there are some typos in your last commit: https://github.com/jgaeddert/liquid-dsp/commit/a9bdcba15b6688040366f1b3d0b11bfdf68297be

JayKickliter, 2015-06-02:
Thanks again for this. I've started working on a [Julia](http://julialang.org) [wrapper](https://github.com/jgaeddert/liquid-dsp) for Liquid.