jgaeddert/liquid-dsp #7: ARM support

andresv, 2015-03-01:
Tried to build it for BeagleBoneBlack which has Cortex A8 and NEON support.

Currently https://github.com/jgaeddert/liquid-dsp/blob/master/configure.ac has such hardcoded configuration for ARM:

    arm*)
        # TODO: check for Neon availability

        # ARM architecture : use neon extensions
        MLIBS_DOTPROD="src/dotprod/src/dotprod_cccf.neon.o \
                       src/dotprod/src/dotprod_crcf.neon.o \
                       src/dotprod/src/dotprod_rrrf.neon.o \
                       src/dotprod/src/sumsq.o"
        # TODO: check these flags
        ARCH_OPTION="-ffast-math -mcpu=cortex-a8 -mfloat-abi=softfp -mfpu=neon";;
    *)

For BeagleBoneBlack

    ARCH_OPTION="-ffast-math -mcpu=cortex-a8 -mfloat-abi=softfp -mfpu=neon";;
must be changed to

    ARCH_OPTION="-ffast-math -mcpu=cortex-a8 -mfloat-abi=hard -mfpu=neon";;

Not sure what is correct for new 4 core RaspberryPi. I got some hints that this should work with gcc >= 4.9

    ARCH_OPTION="-ffast-math -mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4";;

As comments are already saying, question is how to detect if NEON is supported and which neon flag to use, the same goes for mcpu flag.

epsilon519, 2015-03-13:
I can confirm that this work on the RPi 2 B+ but only if you use **GCC 4.8**

```ARCH_OPTION="-ffast-math -mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4";;```

If you are using gcc 4.6 which is default on raspbian you have to use the following

```ARCH_OPTION="-ffast-math -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard```

jgaeddert, 2016-02-26:
@andresv Are you still having trouble with this? I know it's been a while since you opened this, but gcc 4.9+ seems to resolve this on ARM... or at least on the ARM devices I have tried. I don't have a BBB, so I can't speak to that, unfortunately.

andresv, 2016-02-27:
I have also moved on from BBB. I guess it can be closed as "works for me".

jgaeddert, 2016-02-27:
Ok, thanks. I apologize for dragging my feet on this. Build issues on specific platforms are sometimes hard to reproduce. I'm looking for build environments that don't use auto-tools, but I haven't found anything good.

closed 2016-02-27 by jgaeddert

ArtemPisarenko, 2016-08-14:
Issue wasn't resolved properly.
For example, current configuration causes 'Illegal instruction' run-time crash on ZedBoard platform (Cortex-A9) with Ubuntu 14.04 and gcc 4.9. In order to make it finally work (i.e. to succeed with not only `make`, but also with `make check`, `make bench` and integration with other external projects), you have to use `ARCH_OPTION="-ffast-math -mcpu=cortex-a9 -mfloat-abi=hard -mfpu=neon-fp16";;`
This article describes platform-specific choices: [ARM Cortex-A Processors and GCC Command Lines](https://community.arm.com/groups/tools/blog/2013/04/15/arm-cortex-a-processors-and-gcc-command-lines)
As far as I understand, current build environment doesn't allow to detect/check appropriate options. So the best solution is to use safe variant by default and provide `./configure` options for end user.

jgaeddert, 2016-08-17:
Thanks for the tip, Artem. I'll try to dig into it when I get a chance; I don't have a zedboard at home, though, so testing this out would be tricky for me.