jgaeddert/liquid-dsp #8: libfec \o/

jettero, 2015-06-30:
I see the docs (and the bench output) mention libfec, but I can't for the life of me find libfec — well, more precisely, I found a couple things that might be libfec.

I forked one from Google code, and the more likely one seems to be from ka9q.net … but it won't compile on my modern computer.  Some of the preprocessor directives comment out too much code, which is fixable, but other linking issues come up too.

Is that the right one?  I can probably fix it or is it the Google code one?  Asking for a quick doc fix here, sortof along the lines of changing “libfec” to “libfec by so-and-so.”

jgaeddert, 2015-06-30:
Paul, good point. Yes, I'm referring to Phil Karn's (KA9Q's) old, old libfec library. Unfortunately the assembly is old and doesn't support modern architectures. The easy fix is to force it to use portable C. It's of course a lot slower than the MMX/SSE assembly version he provides, but it gets the job done in a pinch. I'm trying to spin my own decoders (Viterbi, RS, BCH, and LDPC) but it's not high on my priority list.

Updating the documentation is!

jettero, 2015-07-01:
Hi, thanks.  I have no idea what you just said regarding decoders, but perhaps it'll make amusing wiki queries later.  I don't particularly even need fec, it just can't handle building the package incompletely and then having 'make bench' complain to me.

I'll also look for the portable C stuff.  Is it maybe a configure flag?  Thanks again.

jgaeddert, 2015-07-04:
I don't think there's a configure option for it. I can send you a patch that might help.

djbarbalato, 2015-07-08:
How did you force it to use Portable C?  Just modifying the Makefile.in to use something like pcc instead of gcc?

jettero, 2015-07-08:
Oh, I see the patch.  Missed that.

jettero, 2015-07-08:
that patch does not apply cleanly — but fec compiled …

jettero, 2015-07-08:
Well it compiles, but fails to provide cpu_mode() and various other failures when you make test or try to build with it from liquid-dsp.  I can just chalk this up as: doesn't work.  I definitely don't actually need it.

jgaeddert, 2015-07-08:
Ok. I'll try to generate a good patch and share it.

brian-armstrong, 2016-02-23:
@jettero I have a clone which works on x86-64 darwin. It might work for you. https://github.com/brian-armstrong/libfec

Note that it is not clever enough to decide system it is building on, so you may need to invoke `./configure` with `--build` e.g. `./configure --build=x86_64-apple-darwin15.0.0`. There's probably a good way to avoid that but my configure-foo isn't very strong.

jgaeddert, 2016-02-28:
Brian's fix to libfec is better than any patch I could have created.

closed 2016-02-28 by jgaeddert