jgaeddert/liquid-dsp #13: Hard limit? "error: msresamp2_crcf_create(), number of stages should not exceed 10"

cjcliffe, 2015-09-14:
I was wondering if this is a soft limit just for sanity or if it's possible to uncap it..  Or should I really be doing something different like another fixed resampler stage before it?

It happens now when my input bandwidth is 3.2M and I'm attempting to resample to ~1.5khz or lower.  It isn't much of a problem at the moment since the usable bandwidth for devices is below the 10 stage limit -- but I'm about to add support for devices that may go as high as 8M - 10M input bandwidth and it will become a much bigger issue.

jgaeddert, 2015-09-30:
cj, yeah, that limit was imposed just for sanity. The buffer sizes can blow up pretty quickly and I just guess that 2^10 was a reasonable limit. I bumped the limit up to 16 stages (64k resampling rate) in df5a459fa05db. I would suggest, though, that if you're trying to digitize multiple channels, there is a more efficient way to do this than a decimation stage for each one.

Also, sorry for the delayed response. Github isn't wanting to notify me of issues or pull requests for some reason.

closed 2015-09-30 by jgaeddert

cjcliffe, 2015-09-30:
@jgaeddert thanks; I've assumed there's probably a better way to do it and it's been on the list for awhile -- can you point me in the direction of a function or example for something like getting the desired freqdem output rate (i.e. 12k - 200k) without heavy decimation on the I/Q input (i.e. 3-20M)?

jgaeddert, 2015-09-30:
If you're trying to receive multiple channels simultaneously the polyphase filter bank channelizer is the way to go.

But for any signal, the basic strategy is to decimate right away to reduce the sample rate so that you have one stream holding all the signals you're interested in. This gives you a stream at a low sample rate which provides a reduced computational burden on your baseband processor.

If you have just one FM signal that has a bandwidth of about 200 kHz, then yes: decimating the input stream down to maybe 500 kHz is the best you can do. If you wanted to share with me what you're trying to do I could probably give you better advice.

cjcliffe, 2015-09-30:
@jgaeddert I'll organize my thoughts and get back to you via e-mail soon -- sounds like I'm on the right path but I may be trying to make things fit too 1:1 with just the arbitrary re-samplers, and wasting a lot of CPU.

I've had on my list for awhile to start dividing the decimation into optimal islands / groups when there's several signals I want.  Right now as you guessed, except for some pre-processing I'm doing 100% of the work for each one. 

I'm not familiar with the polyphase filter bank channelizer -- I'll take a look at the docs now that I know it's somehow applicable; I'm sure that will open another line of questions :)

Thanks!