jgaeddert/liquid-dsp #20: Feature request: GFSK modulation/demodulation support

cyber5tar86, 2015-11-09:
It would be nice if liquid-dsp would be able to modulate/demodulate GFSK signals.
What do you think about it?

jgaeddert, 2015-11-12:
GFSK meaning Gauss frequency-shift keying? I do have a GMSK (Gauss minimum-shift keying) demodulator in liquid-dsp:

https://github.com/jgaeddert/liquid-dsp/blob/master/src/modem/src/gmskdem.c

It is a bit crude, though, and I have a development branch that improves upon this greatly. It's close to being released.

cyber5tar86, 2015-11-12:
Yes!
actually I would like to use your library for the treatment of modulation used in the Digital Mobile Radio protocol, which is also known as 4FSK

jgaeddert, 2015-11-14:
Sure, that should work fine. Do you have specifics on DMR? I know of it, but I'm not too familiar with it. liquid should handle it fine.

cyber5tar86, 2015-11-14:
Yes! the specs are publicly available; my goal is to modify DSD tool and make use of liquid-dsp library

jgaeddert, 2015-11-21:
Looking at the DMR specs [http://www.etsi.org/deliver/etsi_ts/102300_102399/10236101/02.02.01_60/ts_10236101v020201p.pdf](http://www.etsi.org/deliver/etsi_ts/102300_102399/10236101/02.02.01_60/ts_10236101v020201p.pdf) (section 10.2 "Modulation", page 111), DMR uses 4FSK with a square-root raised cosine filter and a maximum deviation

    D = 3h / 2T

This should be fairly straightforward to demodulation using liquid-dsp's `cpfskdemod` object. I'm going to close this issue, but I'd be happy to help you with this if you want. It would be especially great if you had some captures of some of the bursts to test out on.

closed 2015-11-21 by jgaeddert

cyber5tar86, 2015-11-23:
Of course! Here is the share link http://wikisend.com/download/564134/gqrx_20151123_102213_-29536037_8000000_fc.raw. (~85MB)

jgaeddert, 2015-11-25:
Do you have any metadata about the file? That is, sample rate, center frequency, formatting, etc. For the moment I'm assuming it's 16-bit I/Q format, but it looks as though it might just be real samples?

cyber5tar86, 2015-11-25:
I've recorded it by means of GQRX, as I can remember the sample rate is 8 Ms and the center freq is 162.6875 MHz

jgaeddert, 2015-11-25:
16-bit I/Q samples?

cyber5tar86, 2015-11-26:
Nome,  iq data are saved as 32bit float samples

cyber5tar86, 2016-03-09:
Do you have good news for me about DMR Decoding?

jgaeddert, 2016-03-11:
The  `cpsfskdemod` object should work for you. Here's a program to get you started. This down-converts to baseband, resamples at a rate 1/20 (although you can use whatever rate you need), and then saves the output to a file for plotting.

Shown below is the result when running through Octave:

[image]

Here is the program:

    // cyberstar.c : process DMR sample set and reduce rate #include <getopt.h>
    #include <math.h>
    #include <stdio.h>
    #include <stdlib.h>
    #include <string.h>
    #include <liquid/liquid.h>

    int main(int argc, char*argv[])
    {
        char filename_input[256] = "data/cyber5tar86/gqrx_20151123_102213.raw";
        const char  *filename_output = "cyberstar.m";

        // read input file or use default
        char * filename = argc == 2 ? argv[1] : filename_input;
        FILE * fid0 = fopen(filename,"r");

        // estimate spectrum
        unsigned int nfft  = 4200;
        float        alpha = 1e-4f;
        float        psd[nfft];
        spgramcf periodogram = spgramcf_create_kaiser(nfft, nfft/2, 2.0f);

        // down-convert signal to baseband
        float fc = 0.225f;
        nco_crcf nco = nco_crcf_create(LIQUID_VCO);
        nco_crcf_set_frequency(nco,2.0f*M_PI*fc);

        // resample
        float rate = 0.05f;  // resample rate
        msresamp_crcf resamp = msresamp_crcf_create(rate, 60.0f);

        // open output file
        FILE * fid1 = fopen(filename_output,"w");
        fprintf(fid1,"%% %s : auto-generated file from %s\n", filename_output, __FILE__);
        fprintf(fid1,"clear all;\n");
        fprintf(fid1,"close all;\n");
        fprintf(fid1,"nfft = %u;\n", nfft);
        fprintf(fid1,"x    = zeros(1,200e3);\n");

        // While there are frames in the input file, read them, process
        // them and write them to the output file.
        unsigned int num_samples_input  = 0;
        unsigned int num_samples_output = 0;
        while (!feof(fid0))
        {
            float complex x;
            if ( fread(&x, sizeof(float complex), 1, fid0) != 1)
                break;
            num_samples_input++;
        
            // mix to baseband
            float complex y;
            nco_crcf_mix_down(nco, x, &y);
            nco_crcf_step(nco);

            // resample (decimate); because the decimation rate is so low,
            // this either writes 0 or 1 sample
            float complex z;
            unsigned int num_written;
            msresamp_crcf_execute(resamp, &y, 1, &z, &num_written);

            // accumulate power spectral density estimate
            //spgramcf_accumulate_psd(periodogram, &y, alpha, 1);
            if (num_written > 0) {
                spgramcf_accumulate_psd(periodogram, &z, alpha, 1);

                // dump samples to output file (limit to 200,000)
                if (num_samples_output < 200000)
                    fprintf(fid1,"x(%u) = %12.4e + 1i*%12.4e;\n", num_samples_output+1, crealf(z), cimagf(z));
                num_samples_output++;
            }
        }
        // done with input file
        fclose(fid0);
        
        // write accumulated PSD
        spgramcf_write_accumulation(periodogram,psd);
        printf("read   %9u samples\n", num_samples_input);
        printf("output %9u samples\n", num_samples_output);

        // destroy temporary object
        msresamp_crcf_destroy(resamp);
        nco_crcf_destroy(nco);
        spgramcf_destroy(periodogram);

        unsigned int i;
        for (i=0; i<nfft; i++)
            fprintf(fid1,"psd(%6u) = %12.8f;\n", i+1, psd[i]);
        fprintf(fid1,"figure('position',[100 100 600 800]);\n");
        fprintf(fid1,"subplot(2,1,1)\n");
        fprintf(fid1,"  f = [0:(nfft-1)]/nfft - 0.5;\n");
        fprintf(fid1,"  plot(f,psd,'Color',[0 0.3 0.5],'LineWidth',1.5);\n");
        fprintf(fid1,"  psd_min = floor(min(psd)/10)*10;\n");
        fprintf(fid1,"  psd_max =  ceil(max(psd)/10)*10;\n");
        fprintf(fid1,"  axis([-0.5 0.5 psd_min psd_max]);\n");
        fprintf(fid1,"  xlabel('Normalized Frequency [f/F_s]');\n");
        fprintf(fid1,"  ylabel('Power Spectral Density [dB]');\n");
        fprintf(fid1,"  grid on;\n");
        fprintf(fid1,"subplot(2,1,2)\n");
        fprintf(fid1,"  t = [0:(length(x)-1)];\n");
        fprintf(fid1,"  plot(t,real(x),'Color',[0 0.3 0.5],t,imag(x),'Color',[0 0.5 0.3]);\n");
        fprintf(fid1,"  grid on;\n");
        fprintf(fid1,"  legend('real','imag');\n");
        fclose(fid1);
        printf("results written to '%s'\n", filename_output);

        return 0;
    }

greenoaktree, 2017-05-14:
The following function can not find in liquiddsp,why?
spgramcf_create_kaiser
spgramcf_accumulate_psd
spgramcf_write_accumulation

jgaeddert, 2017-06-04:
@greenoaktree These functions were removed in 1.3.0 to help simplify the interface. Here's a stripped down example of how to use the spgramcf object:

    unsigned int nfft        =   1024;  // spectral periodogram FFT size
    spgramcf q = spgramcf_create_default(nfft);

    for (i=0; i<num_samples; i++) {
        // generate sample
        float complex y;

        // push sample through spgram object
        spgramcf_push(q, y);
    }
    // compute power spectral density output
    float psd[nfft];
    spgramcf_get_psd(q, psd);

    // destroy object
    spgramcf_destroy(q);