jgaeddert/liquid-dsp #22: Configure failure

w9ran, 2015-11-23:
Following the described procedure I get the following error running Configure and would appreciate any assistance.   The target is a Raspberry Pi 2.

Thanks, Bob W9RAN

./configure: line 4571: syntax error near unexpected token `no,,'
./configure: line 4571: `AX_GCC_ARCHFLAG(no,,)'
root@raspberrypi:~/Downloads/liquid-dsp# make
make: *** No targets specified and no makefile found.  Stop.

jgaeddert, 2015-11-25:
Bob, what version of liquid are you using? That is, are you using the latest version from github, master branch? Also, what version of gcc?
73, KK4MNZ

w9ran, 2015-11-25:
Thanks for the reply -  I used the version that came from github:   git clone https://github.com/jgaeddert/liquid-dsp

and gcc is:
gcc (Debian 4.6.3-14+rpi1) 4.6.3

I'm learning linux but when issues like this arise I don't know quite how to resolve them, so I appreciate any assistance - 

Bob W9RAN

jgaeddert, 2015-11-25:
I'm sure it's an issue with configure determining how to generate the compiler flags. This is something I've struggled with the GNU make tools for quite some time.

Unfortunately I'm about to leave for the airport and will be gone for a few days. I'll be back early next week and should be able to give you some more attention then.

w9ran, 2015-11-25:
Thanks Joseph, and have a great Thanksgiving!

Bob W9RAN

andrewcooke-isti, 2015-12-02:
i am also seeing this on centos 6.7 (master from github, gcc 4.4.7)

andrewcooke-isti, 2015-12-02:
updating the macro in scripts to the latest version (17) does not help.  if i comment it out, the next failure is on AX_EXT.  it's as though it's not picking up the scripts directory.  soft linking the scripts directory to "m4" does not help.

andrewcooke-isti, 2015-12-02:
phew.  grabbing 1.2.0 tarball and running just configure (or reconf and configure) appears to work.

jgaeddert, 2015-12-02:
At some point configure.ac changed to point to the scripts directory. I am looking into this. Perhaps reverting to an older version will fix this.

jgaeddert, 2015-12-11:
Try replacing the following line in `bootstrap.sh` from

    aclocal

to

    aclocal -I./scripts

andrewcooke, 2015-12-11:
yup.  works just fine.  (this is same person as above, but non-work git account).

(i'm not actually using liquid-dsp any more - not because of this, but because it turned out things were so simple i could just write a couple of loops.  but thanks for fixing this.)

w9ran, 2015-12-13:
Joe -
that enabled me to get over that hurdle, but make fails with a bazillion
errors that I have no clue about - any suggestions would be appreciated:

Bob W9RAN

creating libliquid.a
ranlib libliquid.a
gcc -shared -Xlinker -soname=libliquid.so -o libliquid.so
-Wl,-whole-archive libliquid.a -Wl,-no-whole-archive -lm -lc
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(libliquid.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(libliquid.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(agc_crcf.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(agc_crcf.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(agc_rrrf.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(agc_rrrf.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(cvsd.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(cvsd.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(bufferf.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(bufferf.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(buffercf.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(buffercf.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(channel_cccf.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(channel_cccf.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(dotprod_cccf.neon.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(dotprod_cccf.neon.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(dotprod_crcf.neon.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(dotprod_crcf.neon.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(dotprod_rrrf.neon.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(dotprod_rrrf.neon.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(sumsq.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(sumsq.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(equalizer_cccf.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(equalizer_cccf.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(equalizer_rrrf.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(equalizer_rrrf.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(crc.o) does not
/usr/bin/ld: failed to merge target specific data of file libliquid.a(crc.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(fec.o) does not
/usr/bin/ld: failed to merge target specific data of file libliquid.a(fec.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(fec_conv.o) does not
[... 531 lines cut ...]
libliquid.a(vectorf_mul.port.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(vectorf_trig.port.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(vectorf_trig.port.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(vectorcf_add.port.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(vectorcf_add.port.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(vectorcf_norm.port.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(vectorcf_norm.port.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(vectorcf_mul.port.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(vectorcf_mul.port.o)
/usr/bin/ld: error: libliquid.so uses VFP register arguments,
libliquid.a(vectorcf_trig.port.o) does not
/usr/bin/ld: failed to merge target specific data of file
libliquid.a(vectorcf_trig.port.o)
collect2: ld returned 1 exit status
makefile:1236: recipe for target 'libliquid.so' failed
make: *** [libliquid.so] Error 1
root@raspberrypi:~/Downloads/liquid-dsp#

jgaeddert, 2015-12-13:
Bob, I haven't determined the appropriate compiler flags for gcc 4.6 to work with SIMD extensions, but there are a few options for moving forward.

The quick solution is to skip over the SIMD components of the library and use the portable C version. This will make liquid-dsp run a bit slower, but it will compile for you and hopefully get you on your way while I dig into this a little deeper:

    $ make clean
    $ ./configure --enable-simdoverride
    $ make -j4

The alternative solution is to install gcc 4.8. This could take some time but I have gotten it to work on my raspberry pi 2 at home.

w9ran, 2015-12-13:
HI Joe,
Well that worked too, so thanks!   However the next speed bump I think lies
over the boundary into CubicSDR-land, as cmake is looking for openGL and I
don't see that among the dependencies:

CMake Error at
/usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:97
(MESSAGE):
  Could NOT find OpenGL (missing: OPENGL_gl_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:288
(_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-2.8/Modules/FindOpenGL.cmake:153
(FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:103 (find_package)

-- Configuring incomplete, errors occurred!

I appreciate your help with liquidDSP and realize these are uncharted
waters and I'm not an experienced captain ;-)

Bob

w9ran, 2015-12-15:
On 12/13/2015 11:44 AM, Joseph D. Gaeddert wrote:
>  but I have gotten it to work on my raspberry pi 2 at home.

Hi Joe,

Since you mentioned working on the Pi 2 yourself, I thought you might 
find this compilation of ARM CPU benchmarks of interest:

http://pastebin.com/Z66K3QBz

It will be very interesting to see where the PINE A64 and other new ones 
come in.

Bob

jgaeddert, 2016-01-18:
I think moving forward I'm just going to recommend gcc version 4.7 or higher:

    $ ./bootstrap.sh
    $ CC=gcc-4.8 ./configure
    $ make -j4

In the future I will definitely need a better build environment. I'm reluctant to use cmake (for other reasons), but it is definitely on my task list.

closed 2016-01-18 by jgaeddert