jgaeddert/liquid-dsp #24: configure syntax error

tomwbarclay, 2015-12-07:
Trying to configure and build liquid-dsp. When I run configure I get 

./configure: line 4571: syntax error near unexpected token `no,,'
./configure: line 4571: `AX_GCC_ARCHFLAG(no,,)'

As I am relatively inexperienced with this script I cant figure out what the syntax should be. 
I an using Ubuntu of x86 64 bit

tomwbarclay, 2015-12-08:
Further update ...

If edit AX_GCC_ARCHFLAG( no, ,   ) to be  AX_GCC_ARCHFLAG(    )

the run as normal ...

> ./configure  ... eventually get 

configure: creating ./config.status
config.status: creating makefile
config.status: error: cannot find input file: `config.h.in'

jgaeddert, 2015-12-09:
Thanks for pointing this out. I'll take a look at it. I'm trying to re-organize the automake tools for this. There's a lot that needs updating in this regard.

jgaeddert, 2016-02-26:
Sorry for the long delay in responding. I can't reproduce this error. Is it still an issue for you?

tomwbarclay, 2016-02-27:
Joseph,
     not at the moment I have been moved to another project .. If I have 
future issues I will be back in touch ..

Thanks and best regards ... Tom

-- 
Tom Barclay
Barclay eSolutions Ltd
High Coombe
Lyme Road
Axminster
Devon, EX13 5BL

Tel : +44 (0) 1297 639342
Mob : +44 (0) 7506 788554
Email : tom.besl.uk@googlemail.com

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jgaeddert, 2016-02-28:
Thanks, Tom. Feel free to re-open this issue if you need to.

closed 2016-02-28 by jgaeddert

AaronDMarasco-VSI, 2016-10-21:
This problem still exists on CentOS 6.
automake 1.11.1
autoconf 2.63

AaronDMarasco-VSI, 2016-10-21:
The workaround is [here](https://github.com/jgaeddert/liquid-dsp/issues/22#issuecomment-164078610) - you need to tweak `bootstrap.sh`. That allows me to build native CentOS 6. Still working on cross-compilation issues.