jgaeddert/liquid-dsp #29: Debian Error Help!

rgood123, 2016-02-12:
Iam trying to install on debian jessie armhf cant get pass "make comand"
ERROR:
root@/liquid-dsp# make 
gcc -I . -I include -Wall -fPIC -g -O2  -ffast-math -mcpu=cortex-a8 -mfloat-abi=softfp -mfpu=neon   -c -o src/libliquid.o src/libliquid.c
In file included from /usr/include/features.h:398:0,
                 from /usr/include/complex.h:25,
                 from include/liquid.h:78,
                 from src/libliquid.c:27:
/usr/include/arm-linux-gnueabihf/gnu/stubs.h:7:29: fatal error: gnu/stubs-soft.h: No such file or directory
 # include <gnu/stubs-soft.h>
                             ^
compilation terminated.
<builtin>: recipe for target 'src/libliquid.o' failed
make: *** [src/libliquid.o] Error 1

jgaeddert, 2016-02-13:
I've heard of users running across this error before, but I have not been able to reproduce it. What version of gcc are you using?

    gcc -dumpversion

cjcliffe, 2016-02-13:
Reproducible here on the Raspberry Pi 2 with latest Raspbian Jessie -- configure.ac changes in issue #7 solves the problem for me.

rgood123, 2016-02-14:
Checked version of GCC 4.9.2
Arm device is a quad core cortex a9 if this makes a difference?
Replaced "configure.ac" with  https://github.com/jgaeddert/liquid-dsp/blob/master/configure.ac from issue #7 following directions changed line: ARCH_OPTION="-ffast-math -mcpu=cortex-a8 -mfloat-abi=hard -mfpu=neon";; For cortex a8 hardfloat . 
liquid-dsp now compiles 
when i try to install CubicSdr it errors here is the output log.

The system is: Linux - 3.0.36+ - armv7l
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/rock/CubicSDR/build/CMakeFiles/3.0.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags: 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/rock/CubicSDR/build/CMakeFiles/3.0.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/rock/CubicSDR/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec3351230490/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3351230490.dir/build.make CMakeFiles/cmTryCompileExec3351230490.dir/build
make[1]: Entering directory '/home/rock/CubicSDR/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/rock/CubicSDR/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec3351230490.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec3351230490.dir/testCCompiler.c.o   -c /home/rock/CubicSDR/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTryCompileExec3351230490
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3351230490.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec3351230490.dir/testCCompiler.c.o  -o cmTryCompileExec3351230490 -rdynamic 
make[1]: Leaving directory '/home/rock/CubicSDR/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/rock/CubicSDR/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec4175021084/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec4175021084.dir/build.make CMakeFiles/cmTryCompileExec4175021084.dir/build
make[1]: Entering directory '/home/rock/CubicSDR/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/rock/CubicSDR/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec4175021084.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec4175021084.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-3.0/Modules/CMakeCCompilerABI.c
Linking C executable cmTryCompileExec4175021084
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec4175021084.dir/link.txt --verbose=1
/usr/bin/cc     -v CMakeFiles/cmTryCompileExec4175021084.dir/CMakeCCompilerABI.c.o  -o cmTryCompileExec4175021084 -rdynamic  
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Debian 4.9.2-10) 
COMPILER_PATH=/usr/lib/gcc/arm-linux-gnueabihf/4.9/:/usr/lib/gcc/arm-linux-gnueabihf/4.9/:/usr/lib/gcc/arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/4.9/:/usr/lib/gcc/arm-linux-gnueabihf/
LIBRARY_PATH=/usr/lib/gcc/arm-linux-gnueabihf/4.9/:/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../:/lib/arm-linux-gnueabihf/:/lib/:/usr/lib/arm-linux-gnueabihf/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'cmTryCompileExec4175021084' '-rdynamic' '-march=armv7-a' '-mfloat-abi=hard' '-mfpu=vfpv3-d16' '-mthumb' '-mtls-dialect=gnu'
 /usr/lib/gcc/arm-linux-gnueabihf/4.9/collect2 -plugin /usr/lib/gcc/arm-linux-gnueabihf/4.9/liblto_plugin.so -plugin-opt=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper -plugin-opt=-fresolution=/tmp/ccuO3e5c.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass- [... cut ...]
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=shmat   -o CMakeFiles/cmTryCompileExec3003385107.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.0/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec3003385107
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3003385107.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=shmat    CMakeFiles/cmTryCompileExec3003385107.dir/CheckFunctionExists.c.o  -o cmTryCompileExec3003385107 -rdynamic 
make[1]: Leaving directory '/home/rock/CubicSDR/build/CMakeFiles/CMakeTmp'

Thank you for your help.

rgood123, 2016-02-14:
In last post forgot post error.
home/rock/CubicSDR/build# cmake ../ -DCMAKE_BUILD_TYPE=Release
32 bit compiler detected
-- Could NOT find LIQUID (missing:  LIQUID_LIBRARIES LIQUID_INCLUDES) 
-- Copying /home/rock/CubicSDR/font/*.fnt to directory /home/rock/CubicSDR/build/x86
-- Configuring file vera_sans_mono32.fnt
-- Configuring file vera_sans_mono48.fnt
-- Configuring file vera_sans_mono18.fnt
-- Configuring file vera_sans_mono12.fnt
-- Configuring file vera_sans_mono24.fnt
-- Configuring file vera_sans_mono16.fnt
-- Copying /home/rock/CubicSDR/font/*.png to directory /home/rock/CubicSDR/build/x86
-- Configuring file vera_sans_mono18_0.png
-- Configuring file vera_sans_mono16_0.png
-- Configuring file vera_sans_mono32_0.png
-- Configuring file vera_sans_mono24_0.png
-- Configuring file vera_sans_mono48_0.png
-- Configuring file vera_sans_mono12_0.png
-- Copying /home/rock/CubicSDR/icon/CubicSDR.ico to directory /home/rock/CubicSDR/build/x86
-- Configuring file CubicSDR.ico
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIQUID_INCLUDES
   used as include directory in directory /home/rock/CubicSDR
   used as include directory in directory /home/rock/CubicSDR
LIQUID_LIBRARIES
    linked by target "CubicSDR" in directory /home/rock/CubicSDR

-- Configuring incomplete, errors occurred!
See also "/home/rock/CubicSDR/build/CMakeFiles/CMakeOutput.log".

rgood123, 2016-02-14:
Thank you for your help.
Debian jessie GCC version 4.9.2 
My arm device is a cortex a9 rk3188 followed the directions in issue #7 replaced "configure.ac with
 https://github.com/jgaeddert/liquid-dsp/blob/master/configure.ac changed line to (ARCH_OPTION="-ffast-math -mcpu=**cortex-a8** -mfloat-abi=hard -mfpu=neon";;) hardfloat. Ran liquid-dsp# make distclean
recompiled ok, installed cubicSdr than soapyRTLSDR  i am happy to say it is working! thank you for pointing me in the right direction

jgaeddert, 2016-02-25:
Thanks, CJ. I'm upgrading to jessie now so I have a system to test this out on.

jgaeddert, 2016-02-26:
I have jessie installed on the rpi2 and everything builds fine with gcc 4.9.2.

@rgood123 Are you still having trouble, or can I close this issue?

closed 2021-09-21 by jgaeddert