jgaeddert/liquid-dsp #43: firpfb behavior

mndza, 2016-05-09:
I just noticed an odd result when performing the same filtering operation through firdecim and firpfb.
I wanted to be able to adjust the output phase using a timing estimate from my signal, so first I tested the output of firpfb with a fixed phase (0).
From my understanding the output signals should be the same (as a polyphase filter is just a more efficient implementation of FIR filtering + decimation/interpolation). Then I noticed that the current implementation just subsamples the filter coefficients to form M filters and each output sample only uses 1 of them.
Am i missing something? Is this behavior unintended? If so, I can help here.
Keep up the good work!

JayKickliter, 2016-05-10:
PFBs do use only one phase (sometimes called filter arm) per output sample. That's what makes them efficient.

Are you saying that the output didn't look right?

cjcliffe, 2016-05-10:
Worked perfectly here after many experiments; some additional documentation would have been good though ;)

JayKickliter, 2016-05-10:
I'm not the author of liquid, but I have written a multirate DSP library for another language. Despite being fairly simple, It took me forever to understand polyphase filtering. I finally realized that it's almost impossible to describe them with static diagrams. If you want more info feel free to contact me. 

I lied in my previous response. If you want to do arbitrary (non-rational) sample rate conversion, a good way is to pick the two phases that straddle the virtual phase that you want, use both of them to output two samples, and then linearly interpolate between those two outputs.

mndza, 2016-05-11:
Thank you for your replies.
I set up a little test to get a better understanding of what I am dealing with.

```
const unsigned int M = 16;
...
firdecim_crcf C0_fir  = firdecim_crcf_create(M, C0, C0_len);
firpfb_crcf C0_pfb = firpfb_crcf_create(M, C0, C0_len);
```

For each block in float complex x[M]:
```
// FIR decimator - automatically deals with the entire block of size M
firdecim_crcf_execute(C0_fir, x, &y0);
// Polyphase filter - feed samples first, then execute
for (int i = 0; i < M; i++) firpfb_crcf_push(C0_pfb, x[i]);
firpfb_crcf_execute(C0_pfb, 0, &y1);
```

If I analyze both outputs of y0 and y1, I get the following (y0 top, y1 bottom):
[image]

I realized by taking a look at the code that the output of y1 is equivalent to using a subsampled filter, like this:
```
float C0_dec[C0_len / M];
for (int i = 0; i < C0_len / M; i++) C0_dec[i] = C0[i*M];
firdecim_crcf C0_fir2  = firdecim_crcf_create(M, C0_dec, C0_len/M);
```

Please tell me if there is something wrong with this code.

I know polyphase filtering uses only one phase/filter arm per each _input_ sample, but at the output stage you should sum all the filter arms (which contain values computed with previous samples) to obtain the complete result: if you do not do this, you are simply discarding filter coefficients (as seen above). The output of `firpfb_crcf_execute(C0_pfb, 0, &y1)` is just the output of the first filter arm! (filter 0).

The efficient part comes from not having to compute samples that would be discarded on further decimation.

In fact, a similarly efficient implementation in software is to just avoid computing the dot product for each new sample and just compute it every M samples (which is in fact what firdecim is doing! but with a fixed phase of 0).

mndza, 2016-05-13:
After reviewing this, I guess this is not the intended use of `firpfb` after all. I am a bit confused by the code and its purpose: the main issue I noticed is that all the M subfilters share the same input sample window to perform the dot product.

If they used different windows, I could simply manually sum the filter arms every M samples to obtain what I expected, like this:

```
for (int i = 0; i < M; i++) {
  firpfb_crcf_push(C0_pfb, x[i]);
  firpfb_crcf_execute(C0_pfb, i, &y0);
  out += y0;
}
out = 0;
```

However, the odd thing is that in `firpfbch` (the polyphase channelizer), each subfilter uses its own input sample window, which is what I expected here too (difference is the DFT instead of a simple sum).

mndza, 2016-05-19:
Sorry for the chain of messages, but I think I finally understand what is going on.

If I am right, `firpfb` is used in liquid to implement the filtering structure defined in _Multirate Digital Filters for Symbol Timing Synchronization in Software Defined Radios_ by Fredric J. Harris and Michael Rice.

This is not the usual polyphase filter by any means (the difference is that instead of M polyphase filters operating in parallel, only one filter is constructed with M set of weights: that explains the confusion in my last posts). This structure expects the filter coefficients to be an oversampled-by-M version of the desired response.

Now I am seeing what I expected at the output.

Thank you again for your work.

closed 2016-05-19 by mndza