jgaeddert/liquid-dsp #47: Compile Breaks at dotprod_cccf.mmx.c

heath33, 2016-07-01:
I'm running through the build of CubicSDR from CJ Cliffe's project https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux.  I'm having a problem building liquid-dsp by pulling from 

git clone https://github.com/jgaeddert/liquid-dsp
liquid-dsp$ cd liquid-dsp
liquid-dsp$ ./bootstrap.sh
liquid-dsp$ ./configure --enable-fftoverride 
liquid-dsp$ make -j4

The make output is posted below.  I have two laptops sitting side by side with the same error in the same location of the build.  Both are i386 machines with Debian.  One is running Jessie and has an AMD dual core and the other is running Sid and has an Intel dual core.

I can pull the source code http://liquidsdr.org/downloads/liquid-dsp-1.2.0.tar.gz and compile and install on both machines with no issues.  However, this does not seem to bode well for CubicSDR, CubicSDR fails to build and complains about liquid-dsp. https://github.com/cjcliffe/CubicSDR/issues/393

Any help would be appreciated.  And thanks for all the time and effort you've put into the project.

Here's the breakpoint

`$ make -j4
gcc -I . -I include -Wall -fPIC -g -O2 -march=pentium-m     -c -o src/libliquid.o src/libliquid.c
gcc -I . -I include -Wall -fPIC -g -O2 -march=pentium-m     -c -o src/agc/src/agc_crcf.o src/agc/src/agc_crcf.c
gcc -I . -I include -Wall -fPIC -g -O2 -march=pentium-m     -c -o src/agc/src/agc_rrrf.o src/agc/src/agc_rrrf.c
gcc -I . -I include -Wall -fPIC -g -O2 -march=pentium-m     -c -o src/audio/src/cvsd.o src/audio/src/cvsd.c
gcc -I . -I include -Wall -fPIC -g -O2 -march=pentium-m     -c -o src/buffer/src/bufferf.o src/buffer/src/bufferf.c
gcc -I . -I include -Wall -fPIC -g -O2 -march=pentium-m     -c -o src/buffer/src/buffercf.o src/buffer/src/buffercf.c
gcc -I . -I include -Wall -fPIC -g -O2 -march=pentium-m     -c -o src/channel/src/channel_cccf.o src/channel/src/channel_cccf.c
gcc -I . -I include -Wall -fPIC -g -O2 -march=pentium-m     -c -o src/dotprod/src/dotprod_cccf.mmx.o src/dotprod/src/dotprod_cccf.mmx.c
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
/usr/lib/gcc/i686-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
/usr/lib/gcc/i686-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
<builtin>: recipe for target 'src/dotprod/src/dotprod_cccf.mmx.o' failed
make: *** [src/dotprod/src/dotprod_cccf.mmx.o] Error 1
make: *** Waiting for unfinished jobs....
`

Kishaihd, 2016-11-09:
I'm having this same problem too on this new laptop, but I didn't have this issue on a previous machine. :/

brian-armstrong, 2016-11-09:
Hey guys,

Just a guess, but, try ./configure --with-gcc-arch=native

(Then you'll need to run make clean && make)

This option will switch the arch flag from pentium-m to native, which will tell the compiler specifically that you have these SSE functions available, although this does depend on your machine actually having them. Most likely, it does have them, since if it didn't, these sections would get #defined out by configure.

Kishaihd, 2016-11-09:
It still failed on make with the same error. This is the output I'm getting, I'm not using the " -j4 " argument with make though. 

In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
/usr/lib/gcc/x86_64-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
/usr/lib/gcc/x86_64-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
<builtin>: recipe for target 'src/dotprod/src/dotprod_cccf.mmx.o' failed
make: *** [src/dotprod/src/dotprod_cccf.mmx.o] Error 1

Kishaihd, 2016-11-09:
Is there some other dependency we're forgetting? It seems weird that it worked on my older machine with the same version of Ubuntu.

brian-armstrong, 2016-11-10:
I don't think so - the SSE intrinsics are part of the compiler itself. Usually when you see these errors it's because the compiler doesn't think your CPU supports them. The configuration system sets some #defines that turn these features on or off based on availability.

I suspect one of two things is happening, either the configure step is incorrectly detecting SSE features or it's detecting them correctly and not setting the right CFLAGS. My bet is on the latter. GCC is fairly strict about not letting you use SSE unless you ask for it, even if the CPU you're building on supports it.

What happens if you edit your makefile and change the CONFIG_CFLAGS line to `CONFIG_CFLAGS = -g -O2 -msse3`? Make sure *not* to rerun ./configure after this change since that'll replace this line back to its default.

Kishaihd, 2016-11-10:
adding the " -msse3 " made it all work for me! Interestingly, it appears I'm missing the fec library? I've searched around and can't seem to find any specific fec library or anything. (This is for "forward error correction", right?) Any ideas?

brian-armstrong, 2016-11-10:
Interesting! I figured -march=native would have the same effect as -msse3. It sounds like the configure script just needs to add the correct flag when building the makefile, then.

libfec would be http://www.ka9q.net/code/fec/ though again the configure script should have disabled the features that require this. You'll definitely want to have it installed though, it adds important error correction methods.

bratwurzt, 2016-11-12:
Same here. This helped (tnx brian-armstrong):
`CFLAGS="-march=native"  ./configure --enable-fftoverride`

brian-armstrong, 2016-11-12:
Interesting. Why was --enable-fftoverride needed? Did not having it break something?

brian-armstrong, 2016-11-12:
I believe https://github.com/jgaeddert/liquid-dsp/pull/65 will get autoconf to produce a configure file that has the right -msse option.

bratwurzt, 2016-11-12:
To be honest, I've just followed instructions for:
https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux