jgaeddert/liquid-dsp #49: Build fails  from tarball,  gcc version 5.3.1 20160413 (Ubuntu 5.3.1-14ubuntu2.1)

rich-sjt, 2016-07-05:
src/math/src/poly.findroots.c:287:9: error: non-floating-point argument in call to function ‘__builtin_isnan’
         if (isnan(du) || isnan(dv)) {
         ^
src/math/src/poly.findroots.c:287:9: error: non-floating-point argument in call to function ‘__builtin_isnan’
<builtin>: recipe for target 'src/math/src/polyc.o' failed
Thanks for the help

ckuethe, 2016-07-05:
I got this too. Casting du and dv to float made the compiler error go away and "make check" still succeeds (" PASSED ALL 42373 CHECKS")

liquid-dsp-1.2.0$ diff -u src/math/src/poly.findroots.c{.orig,}
--- src/math/src/poly.findroots.c.orig	2016-07-05 16:44:30.094003182 -0700
+++ src/math/src/poly.findroots.c	2016-07-05 16:44:36.018079503 -0700
@@ -284,7 +284,7 @@
 #endif
 
         // adjust u, v
-        if (isnan(du) || isnan(dv)) {
+        if (isnan((float)du) || isnan((float)dv)) {
             u *= 0.5f;
             v *= 0.5f;
         } else {

brian-armstrong, 2016-07-08:
HEAD on master already has a fix for this, for what it's worth

https://github.com/jgaeddert/liquid-dsp/commit/3055eb3da9d0a202c1a975f7db0c8370a09a30bc

jgaeddert, 2016-07-10:
Thanks for the notice. I should just spin a new release and tarball soon as the latest one is woefully out of date.

ckuethe, 2016-07-12:
Yes please.