jgaeddert/liquid-dsp #50: Building liquid-dsp 1.2.0 on Mac OS X with LLVM/Clang

muellermartin, 2016-07-06:
Building the release tarball of version 1.2.0 of liquid-dsp fails when using LLVM/Clang on Mac OS X (documented in troubleshooting section of manual/website). Currently the suggested fix is to use GCC instead (related issue #48), but this is not favored by Homebrew's maintainers (see isse Homebrew/homebrew-core#2664). Therefore I tried to build the latest release with LLVM/Clang and the only issues seems to be the configure script checks for `malloc`, `realloc`, `free`, `memset`, `memmove` in libc and `sinf`, `cosf`, `expf`, `cargf`, `cexpf`, `crealf`, `cimagf`, `sqrtf` in libm:

    checking for malloc, realloc, free, memset, memmove in -lc... no
    configure: error: Need standard c library!

and

    checking for sinf, cosf, expf, cargf, cexpf, crealf, cimagf, sqrtf in -lm... no
    configure: error: Need standard (complex) math library!

Possible fixes:
- Release current master if applicable
- Patch release for liquid-dsp e.g. 1.2.1 using the relevant changes from master

The patch could simply replace [lines 160 to 166](https://github.com/jgaeddert/liquid-dsp/blob/v1.2.0/configure.ac#L161-L166) in `configure.ac` with the [corresponding lines in the `master` branch](https://github.com/jgaeddert/liquid-dsp/blob/master/configure.ac#L80-L99).

If this patch is applied building with LLVM/Clang succeeds and `make check` passes (`PASSED ALL 42373 CHECKS` with `WARNINGS : 34`).

brian-armstrong, 2016-07-26:
👍 for a new release. The current master has a lot of good updates.

muellermartin, 2017-03-26:
Like #48 this is no longer an issue with the current version of liquid-dsp (1.3.0).

closed 2017-03-26 by muellermartin