jgaeddert/liquid-dsp #55: How to build the documentation?

kebno, 2016-07-27:
Is it in a separate repo?

jgaeddert, 2016-08-17:
The documentation is an ugly beast for sure. I maintain the documentation online here: http://liquidsdr.org/doc

Is there a specific reason you want to build the documentation yourself?

kebno, 2016-08-17:
I am offline for long periods of time and I prefer docs that are synced with the source as opposed to just mirroring the website.

jgaeddert, 2016-08-21:
I understand. I split the documentation off the main branch because building it involved installing a LOT of dependencies, and it was getting too complex for most people. Would a pre-compiled .pdf be sufficient?

kebno, 2016-08-22:
I would prefer the html docs as they are easier to navigate because of
their internal links. But if you had the internal links working well
in the pdf, that would be fine.

Thanks.