jgaeddert/liquid-dsp #66: Ubuntu 16 - problem to install

caiotelles, 2016-11-22:
I was following the Installation guide, but when I executed the make I had some issues.

Can anyone help me?

> liquid-dsp$ make
> gcc -I . -I include -Wall -fPIC -g -O2     -c -o src/dotprod/src/dotprod_cccf.mmx.o src/dotprod/src/dotprod_cccf.mmx.c
> In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
> src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
> /usr/lib/gcc/x86_64-linux-gnu/5/include/pmmintrin.h:50:1: **error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch**
>  _mm_addsub_ps (__m128 __X, __m128 __Y)
>  ^
> src/dotprod/src/dotprod_cccf.mmx.c:255:11: **error: called from here**
>          s = _mm_addsub_ps( ci, cq );
>            ^
> In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
> /usr/lib/gcc/x86_64-linux-gnu/5/include/pmmintrin.h:50:1: **error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch**
>  _mm_addsub_ps (__m128 __X, __m128 __Y)
>  ^
> src/dotprod/src/dotprod_cccf.mmx.c:255:11: **error: called from here**
>          s = _mm_addsub_ps( ci, cq );
>            ^
> <builtin>: recipe for target 'src/dotprod/src/dotprod_cccf.mmx.o' failed
> make: *** [src/dotprod/src/dotprod_cccf.mmx.o] Error 1

brian-armstrong, 2016-11-22:
Hi,

I believe this is the same issue as https://github.com/jgaeddert/liquid-dsp/issues/47

caiotelles, 2016-11-22:
Sorry brian, you are correct.

Just to document:

1. edit the makefile
2. change the CONFIG_CFLAGS line and add  -msse3 (CONFIG_CFLAGS = -g -O2 -msse3)
3. save the file and NOT rerun ./configure 
4. run make (now will work 👍 )

Thanks people!!

brian-armstrong, 2016-11-30:
Another, probably more stable option, is to apply https://github.com/jgaeddert/liquid-dsp/pull/65 and rerun autoconf (or bootstrap.sh) which will cause configure to emit the right msse flag into the makefile. Then you won't have to worry about accidentally rerunning configure.

charlieh0tel, 2017-01-08:
./configure CFLAGS="-O -msse3" 

also works.

rocket-pig, 2019-12-19:
2019 Dec, still need the above CFLAGS to compile on Ubuntu 19.04 Lenovo x230.