jgaeddert/liquid-dsp #67: make liquid-dsp with error

w79110007, 2016-11-29:
I run ./bootstrap.sh and ./configure , and they all succeed.

when run make, it shows:
andy@andy-Mac-Laptop:~/liquid-dsp$ make
gcc -I . -I include -Wall -fPIC -g -O2     -c -o src/dotprod/src/dotprod_cccf.mmx.o src/dotprod/src/dotprod_cccf.mmx.c
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
/usr/lib/gcc/x86_64-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
/usr/lib/gcc/x86_64-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
<builtin>: recipe for target 'src/dotprod/src/dotprod_cccf.mmx.o' failed
make: *** [src/dotprod/src/dotprod_cccf.mmx.o] Error 1

w79110007, 2016-11-29:
sorry, it seems to have answer below, I will read and hope to solve it.

corq, 2016-11-30:
Anyone following this error or arriving from Google searches,  see issue 'Ubuntu 16 - problem to install #66'  for steps to resolve.