jgaeddert/liquid-dsp #69: Many compilation errors (debian 8.0)

btrsdr, 2016-12-10:
`src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
src/dotprod/src/dotprod_cccf.mmx.c:225:12: warning: SSE vector return without SSE enabled changes the ABI [-Wpsabi]
     __m128 sum = _mm_setzero_ps(); // load zeros into sum register
            ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:115:1: error: inlining failed in call to always_inline ‘_mm_setzero_ps’: target specific option mismatch
 _mm_setzero_ps (void)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:225:12: error: called from here
     __m128 sum = _mm_setzero_ps(); // load zeros into sum register
            ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:929:1: error: inlining failed in call to always_inline ‘_mm_loadu_ps’: target specific option mismatch
 _mm_loadu_ps (float const *__P)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:240:11: error: called from here
         v = _mm_loadu_ps(&x[i]);
           ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:922:1: error: inlining failed in call to always_inline ‘_mm_load_ps’: target specific option mismatch
 _mm_load_ps (float const *__P)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:243:12: error: called from here
         hi = _mm_load_ps(&_q->hi[i]);
            ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:922:1: error: inlining failed in call to always_inline ‘_mm_load_ps’: target specific option mismatch
 _mm_load_ps (float const *__P)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:244:12: error: called from here
         hq = _mm_load_ps(&_q->hq[i]);
            ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:193:1: error: inlining failed in call to always_inline ‘_mm_mul_ps’: target specific option mismatch
 _mm_mul_ps (__m128 __A, __m128 __B)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:247:12: error: called from here
         ci = _mm_mul_ps(v, hi);
            ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:193:1: error: inlining failed in call to always_inline ‘_mm_mul_ps’: target specific option mismatch
 _mm_mul_ps (__m128 __A, __m128 __B)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:248:12: error: called from here
         cq = _mm_mul_ps(v, hq);
            ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:741:1: error: inlining failed in call to always_inline ‘_mm_shuffle_ps’: target specific option mismatch
 _mm_shuffle_ps (__m128 __A, __m128 __B, int const __mask)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:251:12: error: called from here
         cq = _mm_shuffle_ps( cq, cq, _MM_SHUFFLE(2,3,0,1) );
            ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:181:1: error: inlining failed in call to always_inline ‘_mm_add_ps’: target specific option mismatch
 _mm_add_ps (__m128 __A, __m128 __B)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:258:13: error: called from here
         sum = _mm_add_ps(sum, s);
             ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:41:0:
/usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:971:1: error: inlining failed in call to always_inline ‘_mm_store_ps’: target specific option mismatch
 _mm_store_ps (float *__P, __m128 __A)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:276:5: error: called from here
     _mm_store_ps(w, sum);
`

brian-armstrong, 2016-12-10:
Hi @btrsdr 

Try the solution to https://github.com/jgaeddert/liquid-dsp/issues/67 https://github.com/jgaeddert/liquid-dsp/issues/66 or https://github.com/jgaeddert/liquid-dsp/issues/47 and see if that helps

btrsdr, 2016-12-12:
Thanks for help!

closed 2016-12-12 by btrsdr