jgaeddert/liquid-dsp #72: ./configure fails while "checking whether mmx is supported"

Kishaihd, 2016-12-29:
Exact output was:
checking whether mmx is supported... ./configure: line 4843: 0xunknown: value too great for base (error token is "0xunknown")

I'm running this on x86_64 Arch Linux. No other previous errors/warnings were given. I've gone through this process on Ubuntu and Fedora with success, but I haven't run into this problem before. Any clues as to what's going on and what to do? Thanks in advance!

brian-armstrong, 2016-12-29:
Hi @Kishaihd 

I ran into this when targeting Android's x86_64 platform. I believe this fixed in that case https://github.com/jgaeddert/liquid-dsp/pull/62/files#diff-4bb328fbe3180ab54cff2f63f71c3a77

You might want to try that entire PR and see if it fixes it

Kishaihd, 2016-12-31:
Hi @brian-armstrong 

I tried that entire PR, but got the same error. :/ 
The successes I had with Ubuntu and Fedora were on this same laptop, so the hardware is all the same too.

brian-armstrong, 2016-12-31:
Just to be clear, did you invoke bootstrap.sh again after applying the patch?

Kishaihd, 2016-12-31:
I did, I completely removed the liquid-dsp folder, re-pulled that specific PR from github, and started again with ./bootstrap.sh.