jgaeddert/liquid-dsp #73: gcc -Wall -O2 -pthread -lm -lc -lliquid -o benchmark_threaded benchmark_threaded.c

chakpongchung, 2017-01-27:
I am trying this file:

http://liquidsdr.org/blog/raspberry-pi3-benchmarks/benchmark_threaded.c

cpchung:temp$ gcc -Wall -O2 -pthread -lm -lc -lliquid -o benchmark_threaded benchmark_threaded.c
/tmp/ccLGDpmH.o: In function `process_worker':
benchmark_threaded.c:(.text+0x2d): undefined reference to `firfilt_crcf_execute_block'
/tmp/ccLGDpmH.o: In function `process_create':
benchmark_threaded.c:(.text+0x25d): undefined reference to `firfilt_crcf_create_kaiser'
benchmark_threaded.c:(.text+0x271): undefined reference to `firfilt_crcf_set_scale'
benchmark_threaded.c:(.text+0x29b): undefined reference to `randnf'
benchmark_threaded.c:(.text+0x2b5): undefined reference to `randnf'
/tmp/ccLGDpmH.o: In function `process_destroy':
benchmark_threaded.c:(.text+0x305): undefined reference to `firfilt_crcf_destroy'
/tmp/ccLGDpmH.o: In function `benchmark':
benchmark_threaded.c:(.text+0x453): undefined reference to `roundf'
/tmp/ccLGDpmH.o: In function `main':
benchmark_threaded.c:(.text.startup+0x332): undefined reference to `ceilf'
collect2: error: ld returned 1 exit status
cpchung:temp$ locate liquid.h
/home/cpchung/liquid-dsp/include/liquid.h
/usr/local/include/liquid/liquid.h

jgaeddert, 2017-01-27:
It's definitely a linker error. Some versions of gcc want the arguments to be in a different order and I always forget what's correct. Try:

gcc -Wall -O2 -o benchmark_threaded  -pthread -lm -lc -lliquid benchmark_threaded.c

chakpongchung, 2017-01-27:
Hi Joseph,

After removing "-o benchmark_threaded", we still have 4 parameters here " -pthread -lm -lc -lliquid", so 16 combinations. I have tried many of them(not all) but it doesnt seem to work, including the one you suggested. 

My gcc:

cpchung:liquid-dsp$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.4' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 

the test result of "make test":

==================================
 UNSTABLE TESTS:
     57 :    1 warnings fec_v27
     58 :    1 warnings fec_v29
     59 :    1 warnings fec_v39
     60 :    1 warnings fec_v615
     61 :    1 warnings fec_v27p23
     62 :    1 warnings fec_v27p34
     63 :    1 warnings fec_v27p45
     64 :    1 warnings fec_v27p56
     65 :    1 warnings fec_v27p67
     66 :    1 warnings fec_v27p78
     67 :    1 warnings fec_v29p23
     68 :    1 warnings fec_v29p34
     69 :    1 warnings fec_v29p45
     70 :    1 warnings fec_v29p56
     71 :    1 warnings fec_v29p67
     72 :    1 warnings fec_v29p78
     73 :    1 warnings fec_rs8
     79 :    1 warnings fecsoft_v27
     80 :    1 warnings fecsoft_v29
     81 :    1 warnings fecsoft_v39
     82 :    1 warnings fecsoft_v615
     83 :    1 warnings fecsoft_v27p23
     84 :    1 warnings fecsoft_v27p34
     85 :    1 warnings fecsoft_v27p45
     86 :    1 warnings fecsoft_v27p56
     87 :    1 warnings fecsoft_v27p67
     88 :    1 warnings fecsoft_v27p78
     89 :    1 warnings fecsoft_v29p23
     90 :    1 warnings fecsoft_v29p34
     91 :    1 warnings fecsoft_v29p45
     92 :    1 warnings fecsoft_v29p56
     93 :    1 warnings fecsoft_v29p67
     94 :    1 warnings fecsoft_v29p78
     95 :    1 warnings fecsoft_rs8
==================================
 WARNINGS : 34
==================================
 PASSED ALL 84663 CHECKS
==================================

jgaeddert, 2017-01-27:
You can also try compiling and linking in separate steps. First generate the unlinked binary object (using the "-c" flag):

gcc -Wall -O2 -c -o benchmark_threaded.o benchmark_threaded.c

Then link the object the object to the libraries:

gcc -o benchmark_threaded -pthread -lm -lc -lliquid benchmark_threaded.o

The second step is where the ordering matters. The ordering of the libraries together doesn't matter (that is, ordering -lm before -lliquid doesn't make any difference). 

If the above doesn't work, try:

gcc -o benchmark_threaded benchmark_threaded.o -pthread -lm -lc -lliquid

chakpongchung, 2017-01-27:
The second way works!

gcc -Wall -O2 -c -o benchmark_threaded.o benchmark_threaded.c
gcc -o benchmark_threaded benchmark_threaded.o -pthread -lm -lc -lliquid

Thank you!

closed 2017-01-27 by chakpongchung