jgaeddert/liquid-dsp #78: Multi-stage resampler

dom11990, 2017-03-13:
I'm attempting to use the multi stage resampler to resample an input waveform to an arbitrary rate. Unfortunately, I'm not getting the results I would expect. I've looked at the example and am trying to adapt it to my particular use case...

My setup is the following:
Input rate: 37777.0
Output rate: 5540.0
R ~ 0.14665007808984304
As = 120 (i know this is excessive, the problem is present at lower (80) attenuation as well)

My test waveform is a complex tone, amplitude 1, with frequency 55.40 Hz, it has 6819 samples. This should produce ten cycles at the end sample rate and 1000 samples.
 (6819*5540/37777)  =1000.0068..
For the most part, this is what happens. However, the start and endpoints don't look as "nice" as I had hoped. for example, at different "R"s I may get an overshoot of as much as 0.1. Clearly, some is to be expected, I just wanted to verify that this is correct and that I am not doing something wrong.

My code performs three executes, at first, I only push filter delay worth of samples into the resampler. Then I push the remaining input data. This way, I can throw away the useless samples. Lastly I push a set of zeroes (size of delay) into the filter to get out the remaining samples. Then I truncate the output buffer and I'm all done. At this point I have the delay compensated output. Here is the execute portion of the code for reference.

```
    size_t output_size = std::floor(input_data.size() * _r);
    //allocate some buffer space, it will be truncated later, lots of extra space here
    _output_data->resize(output_size + 2 * _delay + std::ceil(_r));
    unsigned int ny = 0;
    size_t written = 0;
    float frac_delay = msresamp_crcf_get_delay(_resamp);
    int delay = std::ceil(frac_delay);
    std::vector<std::complex<float> > shifter;
    shifter.resize(delay, std::complex<float>(0, 0));
    //push the first delay samples through the resampler, we will overwrite the results
    msresamp_crcf_execute(_resamp, input_data.data(), delay, _output_data->data(), &ny);
    //now push the remaining samples through the filter, start at index 0 of the output data
    msresamp_crcf_execute(_resamp, &input_data.at(delay), input_data.size() - delay, _output_data->data(), &ny);
    //lastly, push an array of zeroes with delay size through the filter to get the rest of the samples
    msresamp_crcf_execute(_resamp, shifter.data(), shifter.size(), &_output_data->at(ny), &ny);
    //truncate our output data to only contain the delay compensated result
    _output_data->resize(output_size);
```
Link to some plots:
http://imgur.com/a/k3gwL

My question is quite simple, is this procedure functionally correct? If not, where did I slip up? Thank you for any insight.

Also there is a typo in the documentation: for the multistage resampler interface documentation:
http://liquidsdr.org/doc/msresamp/
The call is:  msresamp_crcf_execute(q,*x,nx,*y,*ny), not msresamp_crcf_filter_execute(q,*x,nx,*y,*ny)

cheers,
dominik

GkvJeep, 2018-12-12:
Hi.
In the documentation http://liquidsdr.org/doc/msresamp/, the last decimator is in the range 0.5 - 1.0, 
how to get a decimation of 3.5.

jgaeddert, 2018-12-15:
@GkvJeep

```
float r = 3.5; // resampling rate (output/input)
float As = 60.0f; // stop-band attenuation
msresamp_crcf resampler = msresamp_crcf_create(r,As);
```