jgaeddert/liquid-dsp #79: Why does fft print so much stuff?  How do I prevent fft_execute from printing?

dhowland, 2017-03-20:
I'm using an embedded application.  I'm working on an embedded application, I don't want all sorts of junk on the screen.  Can anyone explain why I get so much junk printed to the screen when specifying anything other than -n 16 in the example code?  Notice I'm using the -q flag.  I've played with the code, and this is definitely coming from the fft_execute() function.

```
user@host:~/dev/liquid-dsp-1.3.0/examples$ ./fft_example -q -n 10
fft plan [forward], n=10, Cooley-Tukey
10, Cooley-Tukey mixed radix, Q=2, P=5
  2, DFT
  5, DFT
computing 5 DFTs of size 2
i=  0/  5
     45.000000   -45.000000
     -5.000000     5.000000
i=  1/  5
     10.388418    20.388418
     -6.624599     3.375402
i=  2/  5
      1.881910    11.881909
     -8.632713     1.367286
i=  3/  5
     -1.367286     8.632712
    -11.881909    -1.881909
i=  4/  5
     -3.375401     6.624599
    -20.388420   -10.388417
computing 5 DFTs of size 2
i=  0/  5
      0.000004     0.000000
     50.000004   -50.000000
i=  1/  5
      9.999996    -9.999998
     60.000004   -60.000008
i=  2/  5
     19.999996   -19.999996
     70.000000   -70.000000
i=  3/  5
     30.000000   -30.000000
     80.000000   -80.000000
i=  4/  5
     39.999996   -40.000000
     90.000000   -90.000000
rmse =   4.6398e-07
done.
```

dhowland, 2017-03-20:
Well, I think I answered my own question.  It appears this is due to FFT_DEBUG_MIXED_RADIX.

Is this due to --enable-debug?  I just wanted to get debug symbols (-g).

thanks

brian-armstrong, 2017-03-20:
David,

I'm pretty sure -g is always present. You can double check the Makefile to be sure, but I seem to remember always seeing it there

dhowland, 2017-03-20:
I just compiled without --enable-debug, -g is not present.  I will check the makefile to be sure.

brian-armstrong, 2017-03-20:
Ah, nevermind then.

Probably your easiest bet is to modify Makefile.in or run configure and make with CFLAGS="-g" prepended

dhowland, 2017-03-20:
Actually, now that I look, -g isn't there even when --enable-debug is used.

So, my final configure call looks like this:
`CFLAGS="-march=native -g" LDFLAGS="-g" ./configure`

I'm not calling it a bug, but perhaps a change request?

brian-armstrong, 2017-03-20:
Does it compile without -march=native? There's a patch in as of roughly a month ago that I believe fixes that, though you'd need to re-autoconf/bootstrap if you cloned at a prior date

dhowland, 2017-03-20:
i'm behind a proxy and I couldn't get git to work, i downloaded the 1.3.0 tag as an archive.

It does not compile on my machine without -march=native

jgaeddert, 2017-03-26:
I'm digging back in my memory as to why I added this to the configure file. I think because I wanted a relatively simple way to re-compile with debugging flags without having to edit source files, but I think @dhowland has shown this is deprecated. I think the best course of action would be to remove the "#if DEBUG" preprocessor macros in favor of specific "#if DEBUG_SPECIFIC" type macros. Ultimately I should remove all of these macros, but I find them useful every now and then. Thoughts?

dhowland, 2017-03-26:
I think it's fine to have debug prints in your code for when you need to work out a problem.  I don't want to give the wrong impression that I was considering that a bug.  I think the actual problem here is just how the configure script is working.

My expectation, given a moderate history with compiling stuff on Linux, was that `--enable-debug` would include debug symbols in the binaries (`gcc -g`), so I can use GDB.  If I had my way, I'd probably just rename your current `--enable-debug` to `--enable-debug-messages`, and then create a new `--enable-debug` that adds debug symbols to the binaries.

jgaeddert, 2017-03-26:
@dhowland: well, if "DEBUG" is getting defined unexpectedly then it probably is a bug somewhere. I'm (clearly) not an expert on autoconf tools, or compiler flags for that matter. I know many people prefer CMake, and I've looked into porting liquid over to it to help simplify the build process, but there are aspects of CMake that I'm not wild about.

In any case, I like your suggestions. I just pushed a change to use `--enable-debug-messages` and removed `--enable-debug`. Currently autoconf defaults `CFLAGS` to `-g -O2` so GDB symbols in the output binary are enabled by default. I could override this and add the `--enable-debug` option back in to only add `-g` when the user explicitly calls for it.