jgaeddert/liquid-dsp #80: mmx related target specific option mismatch

ZeroChaos-, 2017-03-23:
This might be related to other things that are already reported and being worked on, but I wasn't really sure so I opened a new bug here.

```
x86_64-pc-linux-gnu-gcc -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches -I . -I include -Wall -fPIC -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches     -c -o src/equalization/s
rc/equalizer_rrrf.o src/equalization/src/equalizer_rrrf.c
x86_64-pc-linux-gnu-gcc -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches -I . -I include -Wall -fPIC -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches     -c -o src/fec/src/crc.o 
src/fec/src/crc.c
x86_64-pc-linux-gnu-gcc -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches -I . -I include -Wall -fPIC -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches     -c -o src/fec/src/fec.o 
src/fec/src/fec.c
x86_64-pc-linux-gnu-gcc -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches -I . -I include -Wall -fPIC -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches     -c -o src/fec/src/fec_co
nv.o src/fec/src/fec_conv.c
x86_64-pc-linux-gnu-gcc -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches -I . -I include -Wall -fPIC -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches     -c -o src/fec/src/fec_co
nv_poly.o src/fec/src/fec_conv_poly.c
x86_64-pc-linux-gnu-gcc -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches -I . -I include -Wall -fPIC -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches     -c -o src/fec/src/fec_co
nv_pmatrix.o src/fec/src/fec_conv_pmatrix.c
x86_64-pc-linux-gnu-gcc -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches -I . -I include -Wall -fPIC -Os -mtune=nocona -pipe -ggdb -frecord-gcc-switches     -c -o src/fec/src/fec_conv_punctured.o src/fec/src/fec_conv_punctured.c
In file included from src/dotprod/src/dotprod_rrrf.mmx.c:50:0:
src/dotprod/src/dotprod_rrrf.mmx.c: In function ‘dotprod_rrrf_execute_mmx’:
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.4/include/pmmintrin.h:56:1: error: inlining failed in call to always_inline ‘_mm_hadd_ps’: target specific option mismatch
 _mm_hadd_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_rrrf.mmx.c:203:9: error: called from here
     sum = _mm_hadd_ps(sum, z);
         ^
In file included from src/dotprod/src/dotprod_rrrf.mmx.c:50:0:
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.4/include/pmmintrin.h:56:1: error: inlining failed in call to always_inline ‘_mm_hadd_ps’: target specific option mismatch
 _mm_hadd_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_rrrf.mmx.c:204:9: error: called from here
     sum = _mm_hadd_ps(sum, z);
         ^
make: *** [<builtin>: src/dotprod/src/dotprod_rrrf.mmx.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.4/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
make: *** [<builtin>: src/dotprod/src/dotprod_cccf.mmx.o] Error 1
In file included from src/dotprod/src/sumsq.mmx.c:49:0:
src/dotprod/src/sumsq.mmx.c: In function ‘liquid_sumsqf’:
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.4/include/pmmintrin.h:56:1: error: inlining failed in call to always_inline ‘_mm_hadd_ps’: target specific option mismatch
 _mm_hadd_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/sumsq.mmx.c:85:9: error: called from here
     sum = _mm_hadd_ps(sum, z);
         ^
In file included from src/dotprod/src/sumsq.mmx.c:49:0:
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.4/include/pmmintrin.h:56:1: error: inlining failed in call to always_inline ‘_mm_hadd_ps’: target specific option mismatch
 _mm_hadd_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/sumsq.mmx.c:86:9: error: called from here
     sum = _mm_hadd_ps(sum, z);
         ^
make: *** [<builtin>: src/dotprod/src/sumsq.mmx.o] Error 1
```

brian-armstrong, 2017-03-23:
Are you running the master branch?

ZeroChaos-, 2017-03-23:
1.3.0

ZeroChaos-, 2017-03-23:
Can you please document what cpu features are needed so I can update the deps in gentoo?

And then a release tag if at all possible.

brian-armstrong, 2017-03-23:
It's less to do with features needed than having the right compiler flags. See this patch https://github.com/jgaeddert/liquid-dsp/pull/65

And hopefully Joseph does make a new release with this fix in it :)

ZeroChaos-, 2017-03-23:
Ahh, so it's auto testing for cpu arch and setting the cflags appropriately?  Won't that break binary packages for all distros?

brian-armstrong, 2017-03-23:
I don't know what's typical for distributions to do. Are there still any running computers left on this planet without SSE2? ;)

You can work around this by using standard cross-compile flags for generic x86. I imagine this sort of practice would be fairly standard for someone making a package, but I've never done that specifically so I don't know

ZeroChaos-, 2017-03-23:
It looks like the configure script is testing for support in the current cpu and then enabling the flags.  I have no idea how to use "standard cross-compile flags" to bypass that happening.  I'd be grateful for a lesson though.

brian-armstrong, 2017-03-23:
Hah, I'm not entirely sure either. But I do know that when I've chosen compile target as arm or x86 on android, it doesn't enable the SSE features. But I don't know the nitty gritty of how autoconf actually disables that check :(

brian-armstrong, 2017-03-23:
Oh, oops, misread what you said

I think what you might want is something like ./configure --host=x86_64-pc-linux-gnu-gcc? Looking back at my build scripts I believe --host bypasses local arch testing

ZeroChaos-, 2017-03-23:
In gentoo we typically offer flags the user can set to enable certain cpu features when applicable.  Is there a way to set the flags that I want to rather than let it auto-detect?  If not, I can try to dig up some examples of how it's done now.

brian-armstrong, 2017-03-23:
I have no idea there, unfortunately. I'm not the project owner, just someone enthusiastic about it

You might have some luck checking config.ac to see if it takes any flags like that

jgaeddert, 2017-03-29:
I'm certainly not an expert on detecting compiler options either! If there were a standard method for doing this I don't think it would be a problem, but it seems scripts, package managers, etc. all have different expectations for how to achieve this. I'm open to suggestion. Maybe CMake is all it's cracked up to be?

ZeroChaos-, 2017-03-29:
@jgaeddert afaik, the "correct" way to do this is to have an --enable-mmx type flag for each thing you want to enable rather than attempting to auto-detect.  This way, you can set the default to be autodetect but packagers can override the default with their needed option.  Would that be possible?

jgaeddert, 2017-03-29:
Yes, I think that would be reasonably simple.