jgaeddert/liquid-dsp #90: Linker issues when compiling for ARM target + FEC detection question

ijustlovemath, 2017-05-15:
I'm a Linux dev for a university CubeSat team, and we hope to use your frontend to apply Reed-Solomon encoding to our radio transmissions for FEC.

1) How can we detect the number of bit errors if we don't have the original message present on the receiving end? All of the example programs simply use the original message to do this, but I thought Reed-Solomon has some sort of built in "parity checking" that you can use to gauge if you're past the number of correctable errors.

2) We're compiling libfec for the MitySOM335x and are seeing the following errors:

```
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... no
checking for stdlib.h... (cached) yes
checking for GNU libc compatible realloc... no
```
```
checking fec.h usability... no
checking fec.h presence... no
checking for fec.h... no
checking fftw3.h usability... no
checking fftw3.h presence... no
checking for fftw3.h... no
checking for fftwf_plan_dft_1d in -lfftw3f... no
configure: WARNING: fftw3 library useful but not required
checking for create_viterbi27 in -lfec... no
configure: WARNING: fec library useful but not required

```

Related to not finding malloc:
```
src/audio/src/cvsd.c: In function 'cvsd_create':
src/audio/src/cvsd.c:74:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_golay2412.c: In function 'fec_golay2412_create':
src/fec/src/fec_golay2412.c:240:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_hamming74.c: In function 'fec_hamming74_create':
src/fec/src/fec_hamming74.c:60:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_hamming84.c: In function 'fec_hamming84_create':
src/fec/src/fec_hamming84.c:76:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_hamming128.c: In function 'fec_hamming128_create':
src/fec/src/fec_hamming128.c:146:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_pass.c: In function 'fec_pass_create':
src/fec/src/fec_pass.c:34:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_rep3.c: In function 'fec_rep3_create':
src/fec/src/fec_rep3.c:35:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_rep5.c: In function 'fec_rep5_create':
src/fec/src/fec_rep5.c:35:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_secded2216.c: In function 'fec_secded2216_create':
src/fec/src/fec_secded2216.c:200:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_secded3932.c: In function 'fec_secded3932_create':
src/fec/src/fec_secded3932.c:216:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/fec_secded7264.c: In function 'fec_secded7264_create':
src/fec/src/fec_secded7264.c:219:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/interleaver.c: In function 'interleaver_create':
src/fec/src/interleaver.c:81:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/packetizer.c: In function 'packetizer_create':
src/fec/src/packetizer.c:98:5: warning: implicit declaration of function 'rpl_malloc'
src/fec/src/packetizer.c: In function 'packetizer_realloc_buffers':
src/fec/src/packetizer.c:419:5: warning: implicit declaration of function 'rpl_realloc'
src/filter/src/firdespm.c: In function 'firdespm_create':
src/filter/src/firdespm.c:249:5: warning: implicit declaration of function 'rpl_malloc'
src/framing/src/bpacketgen.c: In function 'bpacketgen_create':
src/framing/src/bpacketgen.c:85:5: warning: implicit declaration of function 'rpl_malloc'
src/framing/src/bpacketgen.c: In function 'bpacketgen_recreate':
src/framing/src/bpacketgen.c:160:5: warning: implicit declaration of function 'rpl_realloc'
src/framing/src/bpacketsync.c: In function 'bpacketsync_create':
src/framing/src/bpacketsync.c:103:5: warning: implicit declaration of function 'rpl_malloc'
src/framing/src/bpacketsync.c: In function 'bpacketsync_reconfig':
src/framing/src/bpacketsync.c:449:5: warning: implicit declaration of function 'rpl_realloc'
src/framing/src/detector_cccf.c: In function 'detector_cccf_create':
src/framing/src/detector_cccf.c:124:5: warning: implicit declaration of function 'rpl_malloc'
src/framing/src/framegen64.c: In function 'framegen64_create':
src/framing/src/framegen64.c:53:5: warning: implicit declaration of function 'rpl_malloc'
[... 87 lines cut ...]

Everything compiles and works fine natively, so it's possible that our CMake setup is not passing options appropriately to autoconf. Here's what we're using to pull in your project:

```
    ExternalProject_Add(${LIQUID_NAME}
        DEPENDS fec
        PREFIX ${LIQUID_NAME}
        GIT_REPOSITORY https://github.com/jgaeddert/liquid-dsp.git
        INSTALL_COMMAND ${CDH_SUDO} ${MAKE} install # ${CDH_INSTALL_CMD}
        INSTALL_DIR ${CDH_INSTALL_DIR}
        CMAKE_ARGS
            -DCMAKE_INSTALL_PREFIX=<INSTALL_DIR> ${CDH_CMAKE_ARGS}
        UPDATE_COMMAND ${LIQUID_LIB_BUILD_DIR}/bootstrap.sh
        CONFIGURE_COMMAND ${LIQUID_LIB_BUILD_DIR}/configure --prefix=<INSTALL_DIR> ${CDH_AUTOCONF_HOST} ${CDH_AUTOCONF_BUILD}
        BUILD_COMMAND ${MAKE}
    ##    CMAKE_ARGS -DCMAKE_INSTALL_PREFIX:PATH=${LIQUID_LIB_PREFIX_DIR}
    ##    CMAKE_GENERATOR ${CMAKE_GENERATOR}
         BUILD_IN_SOURCE 1
    ##    INSTALL_COMMAND ""
    #    UPDATE_COMMAND ""
        )
```
The important bit is the ${CDH_AUTOCONF_HOST} and ${CDH_AUTOCONF_BUILD} variables, which have respective values of `--host=arm-arago-linux-gnueabi` and `--build=i386-pc-linux-gnu`

I think this has something to do with the include directories used by cmake not being properly passed into autoconf, though that's just a guess. Have you all successfully built this library for ARM targets before?

itdaniher, 2017-05-16:
Chiming in - I've built liquid-dsp (for windytan's redsea amongst others) just fine on Ubuntu 16.04 and Debian 8 and 9, on both 32b and 64b ARM platforms.

Cheers!

ijustlovemath, 2017-05-16:
Did you have to do anything special, say, before running the bootstrap
script, or while running the configure script, aside from specifying host
and build?

itdaniher, 2017-05-16:
Not that I recall. What distro are you targetting?

ijustlovemath, 2017-05-16:
It's a custom Busybox distro built using LinuxLink. Though we are using the
vendor provided toolchain. The difference comes right at link time; the
native compiler has a "-L/path/libfec.a" and the vendor provided one does
not. Is this something we could force using the configure script? Or is it
possible this needs to happen during the build step?

ijustlovemath, 2017-05-17:
We fixed this by defining CPPFLAGS and CFLAGS to manually include the headers and static libraries required during the CONFIGURE_COMMAND part of our ExternalProject_Add()

closed 2017-05-17 by ijustlovemath