jgaeddert/liquid-dsp #91: Best way to access "derrors" integer from decode_rs_char()?

ijustlovemath, 2017-05-17:
I noticed that the number of errors detected in the example programs all were in reference to the original data. Where we don't have the original data, it would be really nice to have "derrors" available either through direct access to the struct, or through some sort of helper function on the fec_s struct.

Thoughts?

jgaeddert, 2017-05-29:
This is a good idea. I've actually wanted to re-write parts of the FEC library in liquid to support features like this (as well as support other coding types). Way back in the day I think I tried something like this. How about changing `void fec_decode(...)` to `int fec_decode(...)`?

ijustlovemath, 2017-05-29:
Sounds good! I also think you could simply expose the number of correctable
symbol errors through something like fec_max_correctable_symbol_errors(),
so on the other end it would be simple to check "if fec_decode() >
fec_max_...() { //failure code here }"

I'm wondering how people deploy fec_decode without being able to check the
number of symbol errors that occurred.

I ended up basically having to make calls to libfec directly to support
this, as libliquid proved a little too complicated for my skill set to
provide a patch.

Here's what I did:

```
//==========================================================
====================

int rsDestroy(FEC_OBJECT *f)
{
    if(f == NULL) return EXIT_FAILURE;

  //  fec_destroy(f->q);

    if(f->sync == FEC_SYNC)
        free_rs_char(f->rs);

    f->sync = ~FEC_SYNC;
    return EXIT_SUCCESS;
}
//==========================================================
====================

int rsInit(FEC_OBJECT *f)
{
    if(f == NULL) return EXIT_FAILURE;

    if(f->sync == FEC_SYNC) return EXIT_SUCCESS; // we've already created
the object
    f->sync = FEC_SYNC;

  //  fec_scheme fs = LIQUID_FEC_RS_M8;   // error-correcting scheme
  //  f->q = fec_create(fs, NULL);

    f->rs = init_rs_char(RS_SYM_LEN
            , RS_GEN_POLY
            , RS_FCS
            , RS_PRIM
            , RS_NROOTS
            , 0 // Never pad, we keep that metadata ourselves
            );
    f->nerrs = 0;

    return EXIT_SUCCESS;
}

//==========================================================
====================

int rsEncode(FEC_OBJECT *f, uint8_t *src, uint8_t *dest)
{
    if(f == NULL) return EXIT_FAILURE;
    if(rsInit(f)) return EXIT_FAILURE;
    // Encode the src buffer into the dest buffer
#if 0
    //fec_encode(f->q, RS_DATA_LEN, src, dest);
#else
    memcpy(dest, src, RS_DATA_LEN);
    encode_rs_char(f->rs, dest, &dest[RS_DATA_LEN]);
#endif
    return EXIT_SUCCESS;
}

//==========================================================
====================

int rsDecode(FEC_OBJECT *f, uint8_t * src, uint8_t * dest)
{
    if(f == NULL) return EXIT_FAILURE;
    if(rsInit(f)) return EXIT_FAILURE;
    // Decode the src buffer into the dest buffer
#if 0
    fec_decode(f->q, RS_MSG_LEN, src, dest);
    void * zeros = calloc(RS_MSG_LEN, sizeof(unsigned char));
    //if(memcmp(zeros, (*f)->derrlocs))
    free(zeros);
#else
    memcpy(dest, src, RS_MSG_LEN);
    int derrlocs[RS_MSG_LEN];
    int nerrs = decode_rs_char(f->rs, dest, derrlocs, 0);
    f->nerrs = nerrs;
    if(nerrs > RS_DET_ERRS) return EXIT_FAILURE;
#endif
    return EXIT_SUCCESS;
}
```

As you can see, the functionality I needed could most likely be easily
patched into libliquid, but it is a core piece that's currently just
missing completely. I also stored the number of errors within the object
itself, to avoid the need to unnecessarily create temporary variables when
I'm not immediately using the number of detected errors.

Let me know what you think!

ijustlovemath, 2017-05-29:
```
a
b
c

```

ijustlovemath, 2017-05-29:
Sounds good! I also think you could simply expose the number of correctable symbol errors through something like fec_max_correctable_symbol_errors(), so on the other end it would be simple to check `if fec_decode() > fec_max_...() { //failure code here }` I'm wondering how people deploy fec_decode without being able to check the number of symbol errors that occurred. I ended up basically having to make calls to libfec directly to support this, as libliquid proved a little too complicated for my skill set to provide a patch. Here's what I did: 
Sounds good! I also think you could simply expose the number of correctable symbol errors through something like fec_max_correctable_symbol_errors(), so on the other end it would be simple to check "if fec_decode() > fec_max_...() { //failure code here }"

I'm wondering how people deploy fec_decode without being able to check the number of symbol errors that occurred. 

I ended up basically having to make calls to libfec directly to support this, as libliquid proved a little too complicated for my skill set to provide a patch.

Here's what I did:

```
typedef struct __attribute__((packed)) {
    uint32_t sync;
#if(USE_LIQUID)
    fec q;
#endif
    struct rs * rs;
    int nerrs;
} FEC_OBJECT;
//==============================================================================

int rsDestroy(FEC_OBJECT *f)
{
    if(f == NULL) return EXIT_FAILURE;

  //  fec_destroy(f->q);

    if(f->sync == FEC_SYNC)
        free_rs_char(f->rs);

    f->sync = ~FEC_SYNC;
    return EXIT_SUCCESS;
}
//==============================================================================

int rsInit(FEC_OBJECT *f)
{
    if(f == NULL) return EXIT_FAILURE;

    if(f->sync == FEC_SYNC) return EXIT_SUCCESS; // we've already created the object
    f->sync = FEC_SYNC;

  //  fec_scheme fs = LIQUID_FEC_RS_M8;   // error-correcting scheme
  //  f->q = fec_create(fs, NULL);

    f->rs = init_rs_char(RS_SYM_LEN
            , RS_GEN_POLY
            , RS_FCS
            , RS_PRIM
            , RS_NROOTS
            , 0 // Never pad, we keep that metadata ourselves
            );
    f->nerrs = 0;

    return EXIT_SUCCESS;
}

//==============================================================================

int rsEncode(FEC_OBJECT *f, uint8_t *src, uint8_t *dest)
{
    if(f == NULL) return EXIT_FAILURE;
    if(rsInit(f)) return EXIT_FAILURE;
    // Encode the src buffer into the dest buffer
#if 0
    //fec_encode(f->q, RS_DATA_LEN, src, dest);
#else
    memcpy(dest, src, RS_DATA_LEN);
    encode_rs_char(f->rs, dest, &dest[RS_DATA_LEN]);
#endif
    return EXIT_SUCCESS;
}

//==============================================================================

int rsDecode(FEC_OBJECT *f, uint8_t * src, uint8_t * dest)
{
    if(f == NULL) return EXIT_FAILURE;
    if(rsInit(f)) return EXIT_FAILURE;
    // Decode the src buffer into the dest buffer
#if 0
    fec_decode(f->q, RS_MSG_LEN, src, dest);
    void * zeros = calloc(RS_MSG_LEN, sizeof(unsigned char));
    //if(memcmp(zeros, (*f)->derrlocs))
    free(zeros);
#else
    memcpy(dest, src, RS_MSG_LEN);
    int derrlocs[RS_MSG_LEN];
    int nerrs = decode_rs_char(f->rs, dest, derrlocs, 0);
    f->nerrs = nerrs;
    if(nerrs > RS_DET_ERRS) return EXIT_FAILURE;
#endif
    return EXIT_SUCCESS;
}
```

As you can see, the functionality I needed could most likely be easily patched into libliquid, but it is a core piece that's currently just missing completely. I also stored the number of errors within the object itself, to avoid the need to unnecessarily create temporary variables when I'm not immediately using the number of detected errors. Let me know what you think!