jgaeddert/liquid-dsp #92: some functions lost!

greenoaktree, 2017-05-18:
The following function can not find in liquiddsp,why?
spgramcf_create_kaiser
spgramcf_accumulate_psd
spgramcf_write_accumulation

jgaeddert, 2017-05-20:
Are you trying to link to the library after installation? or with just building the example programs? What version are you using?

kangqf, 2017-05-24:
I also can't find the function `ofdmflexframegen_writesymbol` in version 1.3.0!

jgaeddert, 2017-05-29:
The interface changed slightly for those in 1.3.0. Use `spgramcf_create_default(nfft)` for default behavior. Rather than running a PSD accumulation, just use `spgramcf_push(q, sample)` or spgramcf_write(q, samples, len)` and which runs the internal accumulation as needed. Then use `spgramcf_get_psd(q, buffer)` to get the output spectrum (size: nfft, values in dB).

jgaeddert, 2017-05-29:
Regarding `ofdmflexframegen_writesymbol`, I simplified the interface in 1.3.0 to allow the user to write an output buffer of *any* length, not just the buffer length. So you can use `ofdmflexframegen_write(q, buf, len)` where `buf` is complex float of length `len` (can be any length).

closed 2017-05-29 by jgaeddert