jgaeddert/liquid-dsp #102: FYI: Windows/C++ Build

brian-armstrong, 2017-08-10:
Just thought I'd show you what changes liquid will need if it's to run on Windows/compile with MSVC. Feel free to close this issue whenever

The big issue is that MSVC does not support C99 float complex, so I ended up aiming to make everything compile as C and as C++ (msvc supports std::complex<float>). But I found out MSVC also doesn't like variable length arrays, so I changed them all to alloca calls. The last major remaining issue was that C++ is pretty strict about mixing floats and doubles, esp std::complex<float> and double.

https://github.com/quiet/quiet-dsp/compare/devel...win#files_bucket

NissimHadar, 2019-07-23:
Does this now build on Windows?  I have found a branch called `win`, but the `devel` branch is required to build `quiet`.

Cheers!

bobasaurus, 2019-09-11:
I'm currently using the windows build of liquid-dsp included with cubicsdr:

https://github.com/cjcliffe/CubicSDR

Clone the repository and it comes with the dll, lib, and header files for msvc in the "external/liquid-dsp" folder.  When writing test code, I just replaced "float complex" with "liquid_float_complex" and it would build in visual studio 2019 without issue.  Have not done a lot of testing yet though... just FFT so far.