jgaeddert/liquid-dsp #104: Illegal instruction

usagi87, 2017-08-30:
Compile works without e.rror but when runnin make check gives illegal instruction error and stops v1.2 compiles without error but ca  nt use it with the software i need.

jgaeddert, 2017-08-31:
version 1.2 is over 5 years old at this point. Is there a reason you can't use the master branch?

usagi87, 2017-09-01:
Sorry for my chaotic question. I meant say v1.3 compiles but when using make check it gives an error illegal instruction and I meant the v1.2 compiles and make check clears without and error, but is too old to be used with the program I need.

Heres error
tests/data/nco_sincos_fsqrt1_2.o src/nco/tests/data/nco_sincos_fsqrt1_3.o src/nco/tests/data/nco_sincos_fsqrt1_5.o src/nco/tests/data/nco_sincos_fsqrt1_7.o autotest/autotestlib.o libliquid.a -o xautotest -lfec -lfftw3f -lm -lc 
./xautotest -v
0: null
1: agc_crcf
makefile:1276: recipe for target 'check' failed
make: *** [check] Illegal instruction

usagi87, 2017-09-01:
little more info on error Program received signal SIGILL, Illegal instruction.
agc_crcf_execute (_q=_q@entry=0xb6008, _x=<optimized out>, 
    _y=_y@entry=0xbefff320) at src/agc/src/agc.c:106
106	    T y2 = crealf( (*_y)*conjf(*_y) );

leelionhdydy, 2017-09-08:
@usagi87

jgaeddert, 2017-09-13:
I can't think what would cause this. I would assume that an illegal instruction would be caught by the compiler. I can't think of anything wrong with that statement. What's your build system? (OS, cpu architecture, etc.)