jgaeddert/liquid-dsp #105: liquid-dsp-1.3.0 does not build, while the git HEAD of today builds

zoobab, 2017-09-07:
Hi,

I wanted to build the 1.3.0 release, and on my gentoo/sabayon box, I have the following error:

```
zoobab@sabayon /home/zoobab/soft/liquid-dsp-1.3.0 []$ make
gcc -I . -I include -Wall -fPIC -g -O2     -c -o src/dotprod/src/dotprod_cccf.mmx.o src/dotprod/src/dotprod_cccf.mmx.c
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
make: *** [<builtin>: src/dotprod/src/dotprod_cccf.mmx.o] Error 1
```
With the GIT head of today, it builds fine though.

jgaeddert, 2017-09-13:
Yes, there was a fix @brian-armstrong made with 8e4ae07f that addressed this. Do you think a 1.3.1 release would help?

zoobab, 2017-09-14:
Yes, it would help, I tried first the release thinking it was more stable and tested. I was wrong. So definitely release a fix for this version.

ZeroChaos-, 2017-10-23:
been more than a month, bumping, new tag please and thank you

jgaeddert, 2017-10-24:
Thanks for the bump and sorry for the delay. Version 1.3.1 tagged and signed.

closed 2017-10-24 by jgaeddert

ZeroChaos-, 2017-10-25:
thanks! looks good right now.