jgaeddert/liquid-dsp #106: AGC functions in documentation aren't in code

DavidQrc, 2017-10-10:
Hi,
I was wondering why the [documentation ](http://liquidsdr.org/doc/agc/)mentions agc squelch functionality, but I don't see the squelch functions in the source code, and my code won't compile when trying to call the functions agc_crcf_squelch_... 

I'm using version 1.3.0.

jgaeddert, 2017-10-11:
I pulled the squelch functionality out of the AGC in order to improve it, but haven't added it back in yet. Is this a critical function for your development? It wouldn't take long to add back in if so. The API will probably change as a result, though.

DavidQrc, 2017-10-13:
It's not critical. I was just investigating it for possible use in an application I'm working on. I've actually developed a squelch function more specific to the application, so no hurry. I was mainly wondering if I was somehow out of sync and the functions were really there.

jgaeddert, 2017-10-13:
Yeah, I'm the one who's out of sync =)

Links:

  1. https://github.com/jgaeddert/liquid-dsp/issues/106#issuecomment-336536949
  2. https://github.com/

DavidQrc, 2017-10-17:
I see that you added the squelch functions back into AGC in this recent commit:

https://github.com/jgaeddert/liquid-dsp/commit/4f401231d098fb34a48eefb53ae2aad71ffb3118.  

Thanks for the updated code!