jgaeddert/liquid-dsp #107: GMSK demodulation starts outputting all zeros

imuli, 2017-10-11:
Eventually, after on the order of a million symbols, the GMSK demodulation stops working and just outputs all zeros. In one case it put out all ones for a little while, then switched to all zeros. The samples I'm using are mostly background noise with occasional packet bursts (obtained with `hackrf_transfer`).

A minimal example of what I'm doing, using samples [here](https://imu.li/tmp/3456.iq.xz):

```c
#include <stdio.h>
#include <stdlib.h>
#include <complex.h>

#include <liquid/liquid.h>

static const float byte_scale = 1.0/128.0f;

static int
get_samp(float complex *s){
	char buf[2];
	int n;
	n = fread(buf, 2, 1, stdin);
	*s = byte_scale * (buf[0] + buf[1] * I);
	return n;
}

int
main(int argc, char **argv){
	const float bandwidth = 1.152e6;
	const float sample_rate = 3 * bandwidth;
	const int samples_per_symbol = sample_rate/bandwidth;
	const float cutoff = bandwidth/sample_rate;

	firfilt_crcf filter = firfilt_crcf_create_kaiser(64, cutoff, 30.0f, 0.0f);
	gmskdem demodulator = gmskdem_create(samples_per_symbol, 8, 0.5);

	unsigned char out = 0;
	int outc = 8;
	float complex s[2];
	while(get_samp(&s[0]) && get_samp(&s[1])){
		// GMSK eventually fails whether or not the low pass filter is applied
		firfilt_crcf_push(filter, s[0]);
		firfilt_crcf_execute(filter, &s[0]);
		firfilt_crcf_push(filter, s[1]);
		firfilt_crcf_execute(filter, &s[1]);

		unsigned sym;
		gmskdem_demodulate(demodulator, s, &sym);

		out <<= 1;
		out |= sym;
		outc--;

		if(outc == 0){
			putc(out, stdout);
			outc = 8;
		}
	}
	gmskdem_destroy(demodulator);
	firfilt_crcf_destroy(filter);

	return EXIT_SUCCESS;
}
```

jgaeddert, 2017-10-11:
Huh. I think it has to do with the adaptive equalizer that I use to try to eliminate inter-symbol interference. If you edit `src/modem/src/gmskdem.c` and change `#define GMSKDEM_USE_EQUALIZER` to `0` this will probably help. In the meantime I should probably use a static equalizer rather than adaptive. My guess is that the adaptive equalizer is getting into some weird state it can't get out of.

F5OEO, 2017-11-25:
Confirm that I have same issue and solved by #define GMSKDEM_USE_EQUALIZER to 0

jgaeddert, 2017-11-26:
Disabled equalization by default with 615a79ab. I'll look into a more stable way to handle this. My apologies for the inconvenience.

closed 2018-02-03 by jgaeddert