jgaeddert/liquid-dsp #108: Any tips for tuning flexframe/ofdmflexframe?

brian-armstrong, 2017-10-18:
Hey Joseph :)

I've been poking at Quiet Modem again recently and I've been trying to overcome an issue I've had since the start. Quiet works quite well when there's a direct path from the transmitter to the receiver but falls apart pretty quick if there are acoustic reflections. Do you have any suggestions on what parts of liquid might be suitable for that issue?

As I understand it, OFDM is supposed to be reasonably decent at handling multipath transmission, which I guess might be a approximately similar to acoustic echoes. What I've found though is that it seems hard to get the receiver to acquire the OFDM signal at all if conditions are marginal. I'm not really sure how to tune parameters to help with this - it looks like Liquid can print Octave plots, but I'm curious if you have any other methods you use for tuning Liquid in noisy/multipath situations.

Cheers and thanks again for making such a great piece of open source software :)
Brian

jgaeddert, 2017-10-19:
Hey, Brian! The acoustic channel is notoriously bad for communications, but you’re right in that ODFM is much better at handling multipath than other communications schemes. I’d be interested in trying to reproduce your results so I can take a closer look at what the issue with detection is. Off the top of my head I can think of a few things that might be an issue (amplifier compression, large amplitude variation across the modulated band, or even just a weak signal).

The best thing for me would be for you to record some samples and send them to me. Since you’re using the audio channel, a .wav file would work, but I’m not sure if that’s easy with your setup.

brian-armstrong, 2017-10-20:
Awesome! Yeah, I could capture some runs. I've found that my setup generally works pretty well if the path is direct, even from several feet away, but any indirection seems to immediately kill transmission.

I'd like to get something super low throughput but highly reliable, since that seems to be a commonly requested use case for Quiet. Even 64 bps would be great if it could survive a couple reflections.

I'll look into making a recording but in the mean time, if you use Chrome, the setup I've been working with is just https://quiet.github.io/quiet-profile-lab with OFDM with 64 subcarriers, PSK2, conv 1/2 9 centered at 18500 Hz and interpolation factor of 40 or so. The constellation looks fairly clean with a line of sight transmission, but it stops acquiring if I try to turn the mic around.

brian-armstrong, 2017-10-20:
I put some printfs in. It looks like I'm reliably getting to the header, but that's when things fall apart. Typical EVM is 0.5-1, RSSI -40 or so, CFO of 0.01. That's actually much better than what I anticipated and suggests that it might be doable, if some of the error can be compensated for. Still need to get a wav capture, though I was thinking of trying to get it post decimation.

jgaeddert, 2017-10-20:
Well, I wouldn't worry too much about post-decimation. I'd be interested in seeing the whole spectrum, even if it's over-sampled by a factor of 40 as you mentioned.