jgaeddert/liquid-dsp #113: freqmodem documentation does not match library

simoninns, 2018-01-01:
The documentation for freqmodem doesn't match the implementation in the library (mentions functions like freqmodem_create which do not exist and seem to be replaced by freqmod_create, etc. which have different parameters).

It would also be great if there was a clear explanation of the FM demodulator with an example of taking a real signal (i.e. carrier x MHz, deviation y MHz from a complex signal returns 'what it returns') to show how to set up and use the module.

jgaeddert, 2018-01-01:
Solid points. Some of the APIs are out of date, I admit. Examples are always good, though.

simoninns, 2018-01-01:
If my report seems terse, it was just my style of 'bug' reporting :)  The library is extremely useful.

I'm currently trying to demodulate an analogue FM signal (I'm not a DSP expert...) and I'm confused by the demod having a complex input and then only providing a real output (with what seems to be half the number of samples).  This was why I started digging into the documentation, but couldn't find a good explanation of the function. 

Thanks for all the great work on the library!

jgaeddert, 2018-01-01:
No worries! I didn't read your report as terse. And I really should be keeping up with API changes in the documentation, so I'm glad you pointed this out to me. I'm sure there are many more.

Pretty much everything in liquid-dsp uses complex baseband representation. This is done for a number of reasons, but the primary one being that it's much easier to represent and manipulate signals in this form. If you have a real-valued input, I suggest you first convert it to complex baseband with a [Hilbert Transform](http://liquidsdr.org/doc/firhilb/) (or by mixing and low-pass filtering).

The reason why the FM demodulator provides a real-valued output is because the information in FM is encoded in the signal's instantaneous frequency which is a real-valued signal.

I assume you're referencing the [freqmodem](http://liquidsdr.org/doc/freqmodem/) documentation on the website?

simoninns, 2018-01-01:
I'm using the hilbert transform to get my sample into complex baseband before sending it to the freqdem_demodulate function.  Perhaps I'm doing it wrong, but the hilbert transform takes 2 real samples for a single complex sample - then I demod it and get 1 real sample per complex sample in - which results in me having only half of the original real samples at the end of the process?

It was the freqmodem documentation that I was referring to and thanks for the prompt responses!

jgaeddert, 2018-01-01:
This is correct. Although you could run the Hilbert transform without down-sampling 2:1 by using the `firhilbf_c2r_execute()` method. If you're having trouble with you program I can certainly help you debug it.

simoninns, 2018-01-01:
Thanks for the offer of help; I shall have another play with it and get back to you if I hit the wall :)  Happy new year!

jgaeddert, 2018-01-01:
Good luck, and happy new year to you too! Shoot me an email (joseph@liquidsdr.org) if you need to send things directly.

simoninns, 2018-01-01:
I have a quick question about the modulation factor:

I have a sample (real) captured at 32MHz.  In the sample is analogue video in the range of 6.75 - 7.9 MHz, FM modulated at 7.1 MHz.  The spec states the FM carrier deviation is +- 50 KHz.

I can see that the modulation factor is  = Frequency deviation / Modulation frequency

But I don't seem to be able to get a number that works...  My calculation gives me a modulation factor of 0.00704225 but I don't get any output with that setting.  I suspect that I might be doing something very wrong :)

jgaeddert, 2018-01-02:
You're on the right track. The modulation factor is actually relative to the sampling rate once the signal has been discretely sampled. So if your signal is sampled at 36 MHz with a center frequency of 7.1 MHz and a bandwidth of about 1.2 MHz, I would first mix your signal down to complex baseband (center it a 0 Hz), and then reduce the sample rate by a factor of maybe 9 so you have a sample rate of just 4 MHz. Then for a 50 kHz deviation you should use a modulation factor of 0.0125.

Keep in mind that the modulation factor is just a multiplier for the output signal to match the input scaling. I've never tried using the `freqdem` object for analog video, but it should work fine.

simoninns, 2018-01-02:
Makes a lot of sense :)  I'll give it a try; thanks for the advice!

simoninns, 2018-01-02:
Sorry to be a pain, but can you tell me the right function to do the 'mix down with a centre frequency of 0'; is that an fft function?  The resampling I can see in the docs, but I don't see the method to shift the frequency.