jgaeddert/liquid-dsp #118: make

nilampradhan2, 2018-02-08:
i want to install liquid-dsp, after i run ./configure i run command "make", it shows error 
make: *** No targets specified and no makefile found.  Stop.

brian-armstrong, 2018-02-08:
That sounds like ./configure failed. Did it print out an error when you ran it?

nilampradhan2, 2018-02-17:
no.
when i run it it display:
Script started, file is ./configure

nilampradhan2, 2018-02-17:
i run it with commond:
sudo script ./configure
am i right?

windytan, 2018-02-17:
You may have just inadvertently overwritten the configure file with the output of the "script" command. 

Don't run "script", just run [these](https://github.com/jgaeddert/liquid-dsp#installation):

    ./bootstrap.sh
    ./configure
    make
    sudo make install

closed 2018-03-10 by jgaeddert