jgaeddert/liquid-dsp #120: cccf tests failing on mac/clang

viraptor, 2018-02-21:
After running the build with default options and `make check` on:
```
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin17.3.0
```
The `qdetector_cccf_linear_n128` test fails, but only when run with other tests:

```
59: qdetector_cccf:
    282 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n64
    283 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n83
    284 : <<FAIL>>  passed    3 /    4 checks ( 75.0%) : qdetector_cccf_linear_n128
    285 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n167
    286 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n256
    287 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n335
    288 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n512
    289 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n671
    290 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n1024
    291 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n1341
    292 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n64
    293 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n83
    294 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n128
    295 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n167
    296 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n256
    297 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n335
    298 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n512
    299 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n671
    300 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n1024
    301 :   PASS    passed    3 /    3 checks (100.0%) : qdetector_cccf_gmsk_n1341
```

When ran on its own it passes:

```
./xautotest -t 284 -v
qdetector_cccf:
  template length (time):   284
  FFT size              :   1024
  detection threshold   :   0.5000
  sum{ s^2 }            :   256.09

frame detected  :   yes
  sample index  :     1023, actual=    1024 (error=      -1)
  gamma hat     :    0.998, actual=   1.000 (error=  -0.002)
  tau hat       :   -0.294, actual=  -0.300 (error=   0.006) samples
  dphi hat      : -0.00032, actual=-0.00000 (error=-0.00032) rad/sample
  phi hat       :  0.52514, actual= 0.50000 (error= 0.02514) radians

    284 :   PASS    passed    4 /    4 checks (100.0%) : qdetector_cccf_linear_n128
==================================
 PASSED ALL 4 CHECKS
==================================
```

I'm happy to provide more details / testing if needed.

jgaeddert, 2018-02-21:
strange. I'll see if I can reproduce it on my end

brian-armstrong, 2018-02-22:
This may be similar to the test fix in https://github.com/jgaeddert/liquid-dsp/pull/101/files

jgaeddert, 2018-02-24:
Confirmed. I added merged changes from @brian-armstrong and am able to reproduce several qdetector checks with various random seed values. On my end I get it with `./autotest -R 1519498834`

jgaeddert, 2018-03-11:
I determined that certain short sequences will fail depending upon what random sequence is generated for the test. I increased the tolerance to 0.25 radians (about 14.3 degrees) in 1d5556511e763; however I should probably adjust the tolerance based on the sequence length. I think this is sufficient for moving forward

closed 2018-03-11 by jgaeddert