jgaeddert/liquid-dsp #125: error: non-floating-point argument in call to function

romeira, 2018-04-04:
I'm trying to build liquid-dsp-fpm version and I get this error while is building:

```
src/math/src/poly.findroots.c: In function ‘polyc_findroots_bairstow_recursion’:
src/math/src/poly.findroots.c:302:9: error: non-floating-point argument in call to function ‘__builtin_isnan’
         if (isnan(du) || isnan(dv)) {
         ^
src/math/src/poly.findroots.c:302:9: error: non-floating-point argument in call to function ‘__builtin_isnan’
<integrado>: fallo en las instrucciones para el objetivo 'src/math/src/polyc.o'
make: *** [src/math/src/polyc.o] Error 1
```

I found some similar error in the issues but not the way to fix it

Thanks in advance by your support.

jgaeddert, 2020-06-19:
I think this was addressed in 3055eb3da9d0a202c1a975f7db0c8370a09a30bc. Merging master into fpm is going to take some considerable work, so for now I just cherry-picked this commit. It builds on passes on my Linux box.

ludovicmartin, 2020-06-19:
It compiles like a charm, thank you!

jgaeddert, 2020-06-20:
Great! I'll work on closing out some of those failing tests.

closed 2020-06-20 by jgaeddert