jgaeddert/liquid-dsp #130: m files are not displayed in Octave

tomwbarclay, 2018-06-07:
I have downloaded the latest liquid-dsp from github and built all the examples. Running modem_example and gmskmodem_example seems to be fine. They both run and I get .m files, so far so good ... 
Load Octave 4.2.2 and select a .m file and run it I get graph window(s) with no content. Has Octave changed or is the .m file output format changed ?  Any help appreciated.

jgaeddert, 2018-06-07:
Running on Octave 4.4.0 does not provide any issues. What specific examples are you running? You might want to change your Octave GNUTERM environment. What do you get when you run the following in Octave?

    getenv('GNUTERM')

tomwbarclay, 2018-06-09:
See below ...

>> getenv('GNUTERM')
ans =
>>

jgaeddert, 2018-06-09:
Can you plot anything? or is it just the example outputs? What happens when you run something like

    plot(randn(1,100))

Does that produce a good plot?

tomwbarclay, 2018-06-12:
I am sure its an Octave issue. I am running it in a VMware Ubuntu 32 bit guest OS. If I load the example  modem_example.m first then it runs OK. If I then load gmskmodem_example.m then I get 2 empty windows plotted. If I then switch back to modem_example.m, it too now produces a blank window.  Rather than waste your time I will pursue this on the Octave support sites.

closed 2018-06-12 by tomwbarclay

jgaeddert, 2018-06-12:
Not a waste of time at all! Just let me know if I can help in any way.

Links:

  1. https://github.com/jgaeddert/liquid-dsp/issues/130#issuecomment-396554381
  2. https://github.com/