jgaeddert/liquid-dsp #131: consider using gcc fmv for dotprod functions

daveol, 2018-07-19:
Right now simd extensions are chosen at compile time for dotprod functions, this can be annoying for packagers. GCC's [fmv](https://gcc.gnu.org/onlinedocs/gcc-6.4.0/gcc/Function-Multiversioning.html) (Function Multiversioning) can do the fuction choice in runtime based on the simd extensions available. [here](https://lwn.net/Articles/691932/) is also a useful LWN article on the matter

JayKickliter, 2018-07-19:
Relevant clang compatibility documentation: https://clang.llvm.org/docs/AttributeReference.html#target-gnu-target

fsheikh, 2018-08-06:
Hi @jgaeddert (and other folks interested in this issue)--

I was intrigued by FMV and started to add support in liquid. The feature seems to work in the sense that we have multiple functions for each architecture without bothering build system for it.
```C
nm src/dotprod/src/dotprod_cccf.o
                 U __cpu_indicator_init
                 U __cpu_model
0000000000000960 T dotprod_cccf_create
0000000000000a00 T dotprod_cccf_destroy
0000000000000a90 T dotprod_cccf_execute
0000000000000a20 T dotprod_cccf_print
00000000000009a0 T dotprod_cccf_recreate
00000000000008c0 T dotprod_cccf_run
0000000000000000 T dotprod_cccf_run4
0000000000000690 t dotprod_cccf_run4.avx.2
0000000000000000 i dotprod_cccf_run4.ifunc
0000000000000460 t dotprod_cccf_run4.mmx.1
0000000000000000 W dotprod_cccf_run4.resolver
0000000000000230 t dotprod_cccf_run4.sse.0
                 U free
                 U _GLOBAL_OFFSET_TABLE_
0000000000000000 r .LC1
0000000000000000 r .LC2
                 U malloc
                 U memcpy
                 U memmove
                 U __mulsc3
                 U __printf_chk
                 U realloc
```
But I have been stopped in tracks (please see above referenced #132) after looking at `dotproduct` benchmark results. 

So basically on my core-i7 x86 machine, I get the following:

| Dotprod     | Plain C                           | FMV                             |AVX                              |
| ------------- |:------------------------------:|:----------------------------:|:----------------------------:|
| cccf_4     |19.11 M t/s, 137.00   c/t | 21.50 M t/s, 124.26   c/t|68.31 M t/s,  39.27   c/t|
|cccf_16|4.84 M t/s, 541.38   c/t|5.47 M t/s, 488.25   c/t|25.52 M t/s, 105.12   c/t|
|cccf_64|1.72 M t/s,   1.52 k c/t|1.73 M t/s,   1.55 k c/t|16.19 M t/s, 165.72   c/t|
|cccf_256|429.71 k t/s,   6.09 k c/t|434.04 k t/s,   6.16 k c/t|4.64 M t/s, 578.61   c/t|

There is a similar pattern for other input data type parameters. In summary it looks like just enabling the *multi-version* feature will not be sufficient in terms of performance. Unless I am missing some really basic thing, the feature does not look worth pursuing (I will be happy to be corrected)
I may try to get my hands on some other targets and collect further results. I can also add ARM support but I currently a lack a target to test on ARM.

JayKickliter, 2018-08-06:
@fsheikh Isn't the point of FMV that you still have to define optimized versions of the function to get any benefit? Looking at your PR, it's just recompiling the same architecturally-generic [function](https://github.com/jgaeddert/liquid-dsp/blob/f25aba90ae5480ca2c19f462ec20d7c51ded1d4d/src/dotprod/src/dotprod.c#L63-L69) for different architectures.

fsheikh, 2018-08-07:
@JayKickliter That was exactly my first thought. Turns out, the first example cited in the LWN article above using `__attribute__((target("x")))` (which supports individual optimization function for each architecture) is only supported by C++ front-end of GCC. I checked this with both GCC 5 and 6. The closest option was `__attribute__(target_clones("x", "y")))` which is supported by GCC6

daveol, 2018-08-07:
``__attribute__((target("x")))`` is not strictly c++, in c it does not resolve automagically though. you will need to write a resolver using ifunc, an [example](/daveol/liquid-dsp/blob/gcc-targets/src/dotprod/src/dotprod_cccf.c#L68). This example is still a work in progress because it only works for x86 so far, not for arm or powerpc.

fsheikh, 2018-08-07:
@daveol right, I avoided writing a resolver since I did not expect so much performance difference between hand crafted routine versus what compiler could do. Anyway good to see your nice work in this direction. I will close my PR

JayKickliter, 2018-08-07:
Too bad. I was hoping for some smart runtime linking.

daveol, 2018-08-07:
```ifunc``` is basically that, the first time you call the function (lets call it A) marked as an ifunc it will run the resolve function (A_resolve for instance), after which the original function (A) gets replaced with the one returned by the resolver (A_sse2 for instance) to speed up subsequent calls to that function (now function A points to A_sse2).

mirh, 2019-12-02:
A pretty [crucial bug](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=82625) was fixed in GCC9. You may want to try again (otherwise [check if](https://hannes.hauswedell.net/post/2017/12/09/fmv/) moving the trick further down the call dispatch tree helps)

fsheikh, 2019-12-02:
@mirh  Thanks for the links, I will give it a try!

fsheikh, 2019-12-03:
...and indeed GCC 9.2 generates multiple [versions](https://godbolt.org/z/qPP49n) of dot-product and a nice resolver hook. Next step update #132 and run some benchmarks.