jgaeddert/liquid-dsp #135: fails make check on Fedora 28 armv7hl

mdomsch, 2018-08-15:
The make check tests fail on Fedora 28 armv7hl architecture.
https://koji.fedoraproject.org/koji/taskinfo?taskID=29089767

./xautotest -v -o autotest.json
0: null
1: agc_crcf
gamma :   0.10000000, rssi :   0.09999982
gamma :   0.30000001, rssi :   0.29999942
gamma : -30.00000000, rssi : -28.99849129
  TEST FAILED: src/agc/tests/agc_crcf_autotest.c line 165 : expected abs((rssi)-(gamma)) (1.00E+00) < tol (1.00E+00)

mdomsch, 2018-08-15:
This may be intermittent or have some other root cause.  We applied a patch to fix up the ppc64le failures, and have not seen this fail on armv7hl since, though we think the changes should be unrelated.

jgaeddert, 2018-08-28:
Thanks for bringing this up. It's quite honestly not a great test, and as you can see the error is very small. The test was written to make sure that the AGC algorithm works correctly, and there wasn't some major logic error introduced by a bad commit or something. I think that's proven out, so maybe running for a longer time to get the AGC to settle might be best.

jgaeddert, 2018-08-29:
Should be resolved with 158b6da14cac8b

closed 2018-08-29 by jgaeddert