jgaeddert/liquid-dsp #142: liquidsdr.org AGC function documentation incorrect

twitchyliquid64, 2018-09-25:
Sorry if this is the wrong spot.

The docs [here](http://liquidsdr.org/doc/agc/) mention:

```
agc_crcf_set_gain_limits(q,gmin,gmax) sets the minimum and maximum gain values, respectively. This effectively specifies e0 and e1 as in [fig-agc-transfer-function] .
```

This function does not exist:

```
# github.com/twitchyliquid64/go-liquiddsp
./agc.go:19:2: could not determine kind of name for C.agc_crcf_set_gain_limits

FAIL	github.com/twitchyliquid64/go-liquiddsp [build failed]
```

Couldnt find a _report_ link on the site for this, so thought I would open a bug here.

jgaeddert, 2018-09-25:
Ah, good catch. Reporting here is fine. You can also email me directly using joseph at liquid sdr dot org.

tomneda, 2024-05-26:
Hello Joseph, 

just stumbled on the same problem here. Needed to limit the maximum gain of the AGC but did not find in the documentation https://liquidsdr.org/doc/agc/ mentioned interface `agc_crcf_set_gain_limits()`. 

Also, I wanted to play around with `agc_crcf_squelch_enable_auto()` but also did not find it in the interface. But I do not really need that.

As you describe the caption **auto-squelch** in such in detail I would expect the code will somehow exists for `agc_crcf_squelch_enable_auto()` or maybe also `agc_crcf_set_gain_limits()`. Maybe I could find it on a side-branch?

noonafter, 2024-11-27:
in AGC(_print) method, "gain%g dB" should ref to "g", instead of "scale"