jgaeddert/liquid-dsp #145: RationalResampler bug

MartinPhNs, 2018-10-26:
The rate is reverse.
If you ask for Decimation=1, Interpolation=3, you would expect the rate to be 3, not 0.3333
If you do _print(), you're displaying 3.
But _get_rate() is returning 0.3333.

in rresample.c,
// Get rate of resampler, **r = P/Q**
float RRESAMP(_get_rate)(RRESAMP() _q)
{
    return **(float)(_q->Q) / (float)(_q->P);**
}

should be, like the comment is saying
  return **(float)(_q->P) / (float)(_q->Q);**

-----------------------
Wrong value being returned in:
unsigned int RRESAMP(_get_interp)(RRESAMP() _q)
return _q->Q;
should be: return _q->P;

-----------------------
Wrong value being returned in:
unsigned int RRESAMP(_get_decim)(RRESAMP() _q)
return _q->P;
should be: return _q->Q;

---------------------------
The example (rresample_crcf_example.c) and the test (rresample_crc_autotest.c) should be updated accordingly.

jgaeddert, 2018-11-16:
Good catch. I'm on it.

jgaeddert, 2018-11-16:
Fixed with d8fb8430

closed 2018-11-16 by jgaeddert