jgaeddert/liquid-dsp #150: Compilation fail on macOS Moyave

RGerzaguet, 2018-12-05:
Dear all,

I am unable to install liquid-dip on my Mac book pro with macOS Moyave 10.14.1
Bootstrap and configure run fine but the make command fail 

```
gcc -g -O2  -msse4.1 -Wall -fPIC -I. -Iinclude  -c -o src/libliquid.o src/libliquid.c
gcc -g -O2  -msse4.1 -Wall -fPIC -I. -Iinclude  -c -o src/agc/src/agc_crcf.o src/agc/src/agc_crcf.c
gcc -g -O2  -msse4.1 -Wall -fPIC -I. -Iinclude  -c -o src/agc/src/agc_rrrf.o src/agc/src/agc_rrrf.c
gcc -g -O2  -msse4.1 -Wall -fPIC -I. -Iinclude  -c -o src/audio/src/cvsd.o src/audio/src/cvsd.c

...
Lot of stuff going well 

and suddenly 
...

libtool: object: src/libliquid.o malformed object (unknown load command 1)
libtool: object: src/agc/src/agc_crcf.o malformed object (unknown load command 1)
libtool: object: src/agc/src/agc_rrrf.o malformed object (unknown load command 1)
libtool: object: src/audio/src/cvsd.o malformed object (unknown load command 1)
libtool: object: src/buffer/src/bufferf.o malformed object (unknown load command 1)
libtool: object: src/buffer/src/buffercf.o malformed object (unknown load command 1)
libtool: object: src/channel/src/channel_cccf.o malformed object (unknown load command 1)
libtool: object: src/dotprod/src/dotprod_cccf.mmx.o malformed object (unknown load command 1)
libtool: object: src/dotprod/src/dotprod_crcf.mmx.o malformed object (unknown load command 1)
libtool: object: src/dotprod/src/dotprod_rrrf.mmx.o malformed object (unknown load command 1)
libtool: object: src/dotprod/src/sumsq.mmx.o malformed object (unknown load command 1)
libtool: object: src/equalization/src/equalizer_cccf.o malformed object (unknown load command 1)
libtool: object: src/equalization/src/equalizer_rrrf.o malformed object (unknown load 
...
````

Any suggestion or fix ? 
Thanks in advance

jgaeddert, 2018-12-07:
I hit this too on my iMac and submitted a fix a couple of days ago which hopefully addresses this. Are you running the latest from master?

RGerzaguet, 2018-12-08:
Thank you very much for your feedback !
 Unfortunately, I am up to date with master branch.

jgaeddert, 2018-12-08:
I updated to 10.14.2 and it still built fine. I'm not an expert at toolchains by any means, but I know there are various flavors of libtool. 
```
$ cat config.log | grep libtool
configure:5420: checking for libtool
configure:5436: found /usr/bin/libtool
configure:5447: result: libtool
ac_cv_prog_ac_ct_LIBTOOL=libtool
LIBTOOL='libtool'
```

For me `configure` finds `/usr/bin/libtool` which is the one installed by Xcode.

RGerzaguet, 2018-12-09:
Thank's a lot for your feedback, it solves the problem by itself.
It seems that I have some dependency breaks due to macports and that libtool was not inherited properly. 
It perfectly works now !
Thank's again for your rapid and constructive feedbacks !
@+R

closed 2018-12-09 by RGerzaguet

pfeatherstone, 2023-10-26:
I get an issue when building on MAC, it doesn't build `libliquid.a`. Instead, it simply installs `libliquid.ar`. Is there a way to tell the configure script to build the static library? cheers