jgaeddert/liquid-dsp #155: problem with firhilbf_decim_execute_block

ashley-b, 2019-03-24:
I am no DSP expert, but I believe there may be a problem with firhilbf_decim_execute_block.
When I push a one real tone tough it I get the complex decimated tone back, but the signals real and imag sign is alternating every output sample. Also in frequncy domain, the positive and negative content is reversed.
Iam 99% sure I have use it correctly.

Also I have used r2c_execute, which works as expected.

I am doing this simulation n Pothosware with https://github.com/ashley-b/PothosLiquidDSP/tree/hilbert_transform

[image]

jgaeddert, 2019-03-25:
I think I can understand your confusion. The "decim" method reduces the sample rate of the input real-valued series by half into a complex one. The problem is how to represent the original time series as a complex baseband. The "real-to-complex" (r2c) conversion doesn't change sample rate, so it just effectively filters out the negative frequency components of the original signal.

But for the decimation process we can only keep half of the spectrum. At first you might think that the half centered around DC would be appropriate, but considering that the Hilbert transform effectively  filters out the negative frequency band, it makes more sense to keep the positive half of the band. This needs to be centered at f=0 to be represented as complex baseband, so it gets shifted down by Fs/4 which is why you see the alternating sign.

This is what it looks like spectrally (from liquid-dsp/example/firhilb_decim_example.c)

[image]

The orange "transformed/decimated" signal is just shifted in the plot to show how it aligns with the original signal.

ashley-b, 2019-03-28:
Thanks for the detailed explanation. That does make sense now.

closed 2019-03-28 by ashley-b