jgaeddert/liquid-dsp #156: Problem decoding lower sideband

righthalfplane, 2019-03-26:
Hi,

   I am trying to decode SSB, but I am having some problems. I have attached code that shows what I do for lower sideband. The problem is that I get poor output where ever I set the frequency (f). On the other hand, if I change the 3000.0's to 6000.0, I get fairly good decoding on upper and lower sideband with f set to the center frequency, but I always receive both with no way to receive just one of them. I have tried a number of different parameters to "ampmodem_create", but the changes have not had any helpful effect. The input sample rate is 50,000 complex sample per second. The output audio rate is 48,000 samples per second. I assume that I am doing something wrong, but I don't know what it is.

    Dale.

[liquid-lower-sideband.txt](https://github.com/jgaeddert/liquid-dsp/files/3006065/liquid-lower-sideband.txt)

jgaeddert, 2019-03-28:
Dale, are you trying to run this on actual data you have collected? Or just simulation?

righthalfplane, 2019-03-30:
I have been listening on CB and several of the amateur bands. With the double band width, I get nearly perfect sound when I put LSB in the bottom half or USB in the upper half of the 6KHZ bandwidth. If I reverse the positions, then I get something strange like the words are being said backwards. I have recorded the raw IQ data (24MB per minute) and I get the same thing when I play it back and decode it.

righthalfplane, 2019-03-31:
I have made the example into a real program. It decodes the SSB and outputs a raw audio file that can be imported into "Audacity" (Signed 16 bit ppm,little-endian,1 channel (mono), Samplerate 48000). I have attached the program and a 30 second IQ file with upper and lower sideband examples. In my program it sounds better because of the agc - I could never get yours to work.  The IQ file can be extracted with tar -zxvf s_rfspace_IQ_27335000_50000_fc.gz. It sounds a little better with 'f' shifted sightly one way or the other off of the channel center. I played the IQ file in sdrGlut and put it out on utube,  so you could see what it sounds like.

https://youtu.be/aJFhwYDgWCM
[s_rfspace_IQ_27335000_50000_fc.gz](https://github.com/jgaeddert/liquid-dsp/files/3026166/s_rfspace_IQ_27335000_50000_fc.gz)
[liquid-lower-sideband.c.txt](https://github.com/jgaeddert/liquid-dsp/files/3026165/liquid-lower-sideband.c.txt)

jgaeddert, 2019-04-02:
Thanks for the examples. I’m tinkering with it to see if I can figure out what’s wrong.

jgaeddert, 2019-04-02:
I wrote a little program to demod the signal you provided:
https://gist.github.com/jgaeddert/846e781dbef25fd396f577d038e7d430

The result is the following audio signal:
[righthalfplane.wav.gz](https://github.com/jgaeddert/liquid-dsp/files/3034262/righthalfplane.wav.gz)

I use [libsoundfile](http://www.mega-nerd.com/libsndfile/) for reading/writing .wav files. It's easy to install and use.

A few thoughts:
  1. You will need to update liquid-dsp for this to build because of a few modifications I made to the AGC (allowing for an output scaling factor to be set)
  1. I think the first few samples of your raw file are corrupt; I have to skip some in order to process it
  1. Your example has two blocks: the first is upper side-band and the second half is lower side-band. I think there is an issue with my AM demod in that it doesn't filter out the off-band properly. I will look into that soon.

righthalfplane, 2019-04-03:
Hi Joseph,

That looks and sounds great.

Some of the SDRs put out bad blocks or drop out altogether for a few cycles, when you change the sample rate or any of a number of things - you just have to recover from the bad data.
 
The IQ sample magnitudes span a wide range - with the netsdr it is around 0.001 to .002, with the  hackrf one, it ranges  0.1 to 2.0. The sdrplay, lime mini and rtlsdr range at various levels between the other two.

The file that I gave you can be played with Gqrx using the Device "Complex Sampled (IQ)" option. Gqrx gives quite good upper and lower sideband seperation.

Dale.

jgaeddert, 2019-04-15:
I pushed some changes (22ccf5fb34a5a4253d984b2f1f81c67b7aa6eab7) that improve SSB demodulation, and includes good USB/LSB isolation using a modified Hilbert transform. The only modification from an API perspective is to remove the "fc" (carrier) argument in the demodulator object as this isn't used any more (see gist https://gist.github.com/jgaeddert/846e781dbef25fd396f577d038e7d430)

You can hear the result from the following wav file which only includes the upper side-band audio:
[righthalfplane.wav.gz](https://github.com/jgaeddert/liquid-dsp/files/3080755/righthalfplane.wav.gz)

closed 2019-04-15 by jgaeddert