jgaeddert/liquid-dsp #157: Documentaion Finite Impulse Response Hilbert Transform (firhilb) typo

ashley-b, 2019-03-30:
http://liquidsdr.org/doc/firhilb/ documentation seams to wrong near the bottom for firhilbf_interp_execute

>  executes the transform as a decimator

 probably should be

> executes the transform as a interpolator

```
firhilbf_decim_execute(q,*x,*y) executes the transform as a decimator, converting a 2-sample input array x→ of real values into a single complex output value y
.
firhilbf_interp_execute(q,x,*y) executes the transform as a decimator, converting a single complex input sample x
into a two real-valued samples stored in the 

                                                              ^^^^^
                                                            interpolator
```

jgaeddert, 2019-04-05:
Good catch. I’m in the process of overhauling the documentation, but catching mistakes like this are very useful. Thanks!

JayKickliter, 2019-04-05:
> I’m in the process of overhauling the documentation

I'm sure other's have told you before, but Liquid has the nicest API docs I've ever seen.

luzpaz, 2024-02-10:
Is this still relevant ?