jgaeddert/liquid-dsp #158: thread safety

pfeatherstone, 2019-04-01:
Don't think liquid is thread safe. Namely, the use of rand() is not thread safe.
I have a version which defines a function rand_safe() that uses a global spinlock which must be initialised at startup. Therefore, there is also a liquid_init() function that initialises the spinlock.
I have replaced rand() with rand_safe() everywhere. I haven't got an official branch since these changes can easily be scripted. Can anyone think of any other thread safety issues with liquid?

brian-armstrong, 2019-04-01:
Just to put my two cents in on this...

This entire library is not designed in a threadsafe way, and that's ok! Being threadsafe in C means pulling in lots of cross-platform primitives to deal with threading, and the full codebase would need a pass to make it threadsafe.

I use liquid in a multithreaded environment. What I've done is make my calling code threadsafe, and it only talks to liquid on one thread. The burden of thread safety and cross platform falls on my code. The burden to make liquid threadsafe would be significant and it's not clear it's worth it.

Also, I'm curious what you're encountering that uses rand(). I know a lot of these tests/utilities do but is it in main modules?

pfeatherstone, 2019-04-01:
Thank you for the reponse.

I am currently writing some signal generation code that mainly uses "symstream" and "channel". I am writing a very large dataset while varying amplitude, phase offsets, etc (all those parameters found in symstream and channel). I'm also adding oscillator noise using Leeson's equation for added variety. I have a very large for-loop which is looping over all those parameters and generating signals. Overall, my dataset is about 1.2 TB. In order to speed things up, I am using OMP to parallelise over 12 threads. In order to make it thread-safe (ish), i had to change rand() to a thread safe version using a global spinlock. I couldn't think of anything else in the code which would require thread safety but posted the issue in case anyone else thought of anything. 

However, I completely get your point. I always think of C libraries as a collection of functions which take inputs and spit out outputs, THAT's it. So I completely agree that thread-safety isn't a necessity. However, in my case, I do care. Adding a mutex or a spinlock over a function call to liquid would defeat the point of using OMP. That's why I haven't created a branch. I'm just hacking away to make it work.

This isn't really an issue, more of a question to be fair.

JayKickliter, 2019-04-05:
> Can anyone think of any other thread safety issues with liquid?

I haven't noticed anything obvious. All the code I've seen appears to be re-entrant, and I don't I've seen a single global. You still have to restrict access to liquid objects to a single thread at a time, but it's reasonable to offload that responsibility to users of the library.

pfeatherstone, 2019-04-15:
Liquid uses FFTW all over the place. Creating FFTW plans isn't thread safe. So that's a big one. I'm gonna just put a recursive mutex lock around any call to liquid

jgaeddert, 2019-04-15:
Thanks for the comments. This is something I've battled with for a while; on the one hand, adding mutexes all over would allow for separate threads to gain access to the same objects without the user worrying about race conditions. This of course makes portability difficult. On the other hand, liquid-dsp did start out as a C library as just a collection of functions as @pfeatherstone said. The issue there was maintaining state, and the list of function arguments got untenable very quickly.

Having said that, I have tried very hard to avoid global variables (aside from constants, which--let's be honest--aren't really *variables*) :)

And while I do use FFTW in many places due to its speed, I do have an internal FFT that can be used in its place. I prefer FFTW in general because it's faster, but the internal one can be used and it should be thread safe.

As for calls to rand(), I do understand that can be an issue. I've considered writing a noise generator object using shift registers, but it unfortunately got pushed down in priority. Certainly wrapping every call to rand() with a spin lock will vastly slow down your signal generation, though.

What would help you most in the immediate future?

pfeatherstone, 2019-04-15:
I have just made my calling code thread safe. I forgot about your internal FFT function.

pfeatherstone, 2019-04-15:
@jgaeddert Might I suggest using a custom pseudo random number generator based on xorshift128 or something. Any object requiring the use of random numbers could use its own instance of a random number generator object. Combined with your own FFT function, and liquid is pretty much thread-safe. Or am I missing something?

closed 2019-04-24 by pfeatherstone

pfeatherstone, 2020-08-17:
Just a few more thoughts. Adding a random number generator object to liquid would also mean that all algorithms using rand() could be reproduced exactly using a custom seed. That's not the case if multiple threads are using rand() at the same time. Regarding FFTs, I don't know how fast liquid's own implementation is, but kissFFT is also an option and the FFT planner is thread safe. kissFFT is not the fastest algorithm in the world, but consists of a handful of files and is trivial to integrate. The only other non-thread safe function I can think of is printf, but no one cares about that let's be honest.