jgaeddert/liquid-dsp #160: Problems on ARM Cortex-A9

beardedN5rd, 2019-04-10:
Hi Joseph,
I am trying to do some rds decoding using https://github.com/windytan/redsea and only get a subset of the messages. Digging deeper I noticed that liquid-dsp have some issues in its checks. I added autotest.json below.
[autotest.json.txt](https://github.com/jgaeddert/liquid-dsp/files/3062778/autotest.json.txt)

Initially with the auto generated build options (CONFIG_CFLAGS   = -g -O2  -ffast-math -mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4) it failed to run due to illegal arguments.
With corrected options (CONFIG_CFLAGS   = -g -O2  -mcpu=cortex-a9 -mfloat-abi=hard -mfpu=neon) it works partly. Most tests run positively, some rds messages get decoded.

My environment is mostly fixed and can not be updated essentially:
- Nvidia Tegra T30 CPU (Cortext A-9, 32bit, neon-vfpv3)
- Ubuntu Trusty (Linux  3.1.10-g8234415e220d #279 SMP PREEMPT Tue Feb 12 11:58:54 CET 2019 armv7l armv7l armv7l GNU/Linux)

Any idea how I can get it work? Do you need some further information?

Hagen

beardedN5rd, 2019-04-10:
Hi Joseph,
maybe it is an issue with NEON on ARMv7-A architecture, like stated in the [gcc options](https://gcc.gnu.org/onlinedocs/gcc-4.8.4/gcc/ARM-Options.html)

> If the selected floating-point hardware includes the NEON extension (e.g. -mfpu=‘neon’), note that floating-point operations are not generated by GCC's auto-vectorization pass unless -funsafe-math-optimizations is also specified. This is because NEON hardware does not fully implement the IEEE 754 standard for floating-point arithmetic (in particular denormal values are treated as zero), so the use of NEON instructions may lead to a loss of precision.

In the meanwhile I tried to use softfp, but got errors as well. 
[autotest.softfp.json.txt](https://github.com/jgaeddert/liquid-dsp/files/3063476/autotest.softfp.json.txt)

jgaeddert, 2019-04-15:
I'd be interested in seeing why those particular (fm demod) tests fail and what the issues are there.

jgaeddert, 2019-04-17:
I was actually going to suggest that! Any samples you could send me would help immensely!