jgaeddert/liquid-dsp #162: Noise generated during decode of Narrow Band FM

righthalfplane, 2019-04-15:
I modified your program to decode Narrow Band FM. I actually sounds quite good - except for the large noise spikes. I have attached the program, IQ file, wav file and the wav file output from another program decoding 10 seconds of the IQ file - it has a minor DC drift problem, but does not show the noise spikes and it has no noise suppression turned on. 

Dale.

[righthalfplaneNBFM.c.gz](https://github.com/jgaeddert/liquid-dsp/files/3082445/righthalfplaneNBFM.c.gz)
[0nbfm_IQ_162390000_50000_fc.wav.gz](https://github.com/jgaeddert/liquid-dsp/files/3082447/0nbfm_IQ_162390000_50000_fc.wav.gz)
[righthalfplane.wav.gz](https://github.com/jgaeddert/liquid-dsp/files/3082450/righthalfplane.wav.gz)
[0nbfm_IQ_162390000_50000_fc.gz](https://github.com/jgaeddert/liquid-dsp/files/3082446/0nbfm_IQ_162390000_50000_fc.gz)

jgaeddert, 2019-04-23:
Are these noise spikes in the RF signal? or in the FM demod output?

righthalfplane, 2019-04-30:
I am not quite sure - I thought that it may be in the rresamp_crcf_create_kaiser routine. When I do things slightly different using some of your routines (msresamp_crcf_execute and msresamp_rrrf_execute) and some of mine, I do not see it (unfortunately it runs at half the speed of this method).  When changed the bw on the rresamp_crcf_create_kaiser filter to the correct value - it cut the noise in half.