jgaeddert/liquid-dsp #163: gcc 5.4 - ubuntu 16 - channel_cccf bug

pfeatherstone, 2019-04-16:
The following code breaks when building liquid with gcc 5.4 on ubuntu 16. I know the advice is to use a later version of gcc but for the sake of completeness, I have posted the issue. Also, it's not the first time i've seen people complaining about gcc 5.4 in other projects. So this could be a gcc 5.4 auto-vectorization bug. Worth knowing anyway.

Here is the code;

#include <cstdlib>
#include <vector>
#include <cmath>
#include <cstdio>
#include <cassert>
#include <complex>
#include <liquid/liquid.h>

using namespace std;

bool cisfinite(float x)
{
    return x == x;
}

bool is_finite(int n, const complex<float>* z)
{
    bool ok = true;
    for (int i = 0 ; i < n && ok; i++)
    {
        ok &= cisfinite(z[i].real());
        ok &= cisfinite(z[i].imag());
    }
    return ok;
}

bool is_bounded(int n, const std::complex<float>* z, float bound)
{
    bool ok = true;
    for (int i = 0 ; i < n && ok; i++)
    {
        ok &= (abs(z[i].real()) < bound);
        ok &= (abs(z[i].imag()) < bound);
    }
    return ok;
}

/*
 * 
 */
int main(int argc, char** argv)
{
    const int n             = 32*1024;
    const float noise_floor = -60.0f;
    const float snr         = 2;
    const int filter        = LIQUID_FIRFILT_ARKAISER;
    const int mod           = LIQUID_MODEM_BPSK;
    const int delay         = 11;
    const float eband       = 0.1;
    const float phi         = M_PI_4;
    const float dphi        = M_PI_4 / n;
    const int mp_ntap       = 3; 
    const float doppler     = 0.1;
    
    vector<complex<float> > buf(n);
    vector<complex<float> > buf2(n);
    
    for (int i = 0 ; i < 1000 ; i++)
    {
        srand(1);
        
        {
            symstreamcf gen = symstreamcf_create_linear(filter, 2, delay, eband, mod);
            symstreamcf_write_samples(gen, &buf[0], n);
            symstreamcf_destroy(gen);
        }

        {
            channel_cccf channel = channel_cccf_create();
            channel_cccf_add_awgn(channel, noise_floor, snr);
            channel_cccf_add_multipath(channel, 0, mp_ntap);
            channel_cccf_add_carrier_offset(channel, dphi, phi);
            channel_cccf_add_shadowing(channel, 1.0f, doppler);
            channel_cccf_execute_block(channel, &buf[0], n, &buf2[0]);
            channel_cccf_destroy(channel);
        }

        for (size_t i = 0 ; i < 20 ; i++)
        {
            printf("z[%lu] = %f + i %f\n", i, buf2[i].real(), buf2[i].imag());
        }
        printf("\n");
        assert(is_finite(n, &buf2[0]));
        assert(is_bounded(n, &buf2[0], 10));
    }
    
    return 0;
}

pfeatherstone, 2019-04-16:
Actually I think it's a genuine Liquid bug. The CHANNEL() object creates a FIRFILT() object for multi-path regardless of whether it will be used. The fir filter object is then recreated before it is used. I think the bug lies in the FIRFILT(_recreate) function (which calls recreate on window and dotprod objects). If you null the FIRFILT() object in the constructor ( CHANNEL() CHANNEL(_create)) and create it normally in the CHANNEL(_add_multipath) function as such:
_q->channel_filter = FIRFILT(_create)(_q->h, _q->h_len); 
then there is no bug. You have to be careful to only free the object if it is not null in the destructor (CHANNEL(_destroy)).

pfeatherstone, 2019-04-16:
Note that you have to repeatedly regenerate the data to reproduce the bug, hence the for-loop. This might point to an array overwrite or memory leak

jgaeddert, 2019-04-16:
can you attach your program as a file rather than copying/pasting? Seems like there is some formatting issues with that.

jgaeddert, 2019-04-16:
I modified the program for C (and removed some error checking); can you try running it and seeing if you see similar results?

https://gist.github.com/jgaeddert/269ebd9fdcf951da10c97d3fee21a6e2

pfeatherstone, 2019-04-16:
The error checking is the crucial bit. Otherwise you won't see the NaNs or the ridiculously large numbers ( > 10^31 )

pfeatherstone, 2019-04-16:
Otherwise the code will run fine. The point is, it's a silent bug. You need the assert statements

pfeatherstone, 2019-04-16:
Also I tested it on ubuntu 18 and gcc 7.3 and found it still broke. I think the bug lies with FIRFILT(_recreate) inside the CHANNEL() object.

pfeatherstone, 2019-04-16:
Also the print statement is there to show that the samples aren't always the same (at least the first few) even though the pseudo random number generator seed is reset to 1.

pfeatherstone, 2019-04-24:
If people are still interested, as a quick fix, if you change the content of FIRFILT(_recreate) with FIRFILT(_destroy) and FIRFILT(_create), you don't get any issues. I know this is not ideal but I can't see any obvious problems. Or, inside CHANNEL(_create), if you null the FIR filter and create it once when you add multi-path, this also fixes it (because you're not calling FIRFILT(_recreate), but simply FIRFILT(_create). I prefer the earlier fix because FIRFILT is used all over the shop and might fix other hidden issues.