jgaeddert/liquid-dsp #165: polynomial fitting

pfeatherstone, 2019-04-20:
I think there is an issue with the polynomial fitting using lagrange interpolation. 
According to the documentation, using polyf_fit and polyf_git_lagrange should be the same if n == k. However I don't get the same answers.
Do other people have the same issues?

`#include <cstdlib>
#include <cstdio>
#include <liquid/liquid.h>

using namespace std;

/*
 * 
 */
int main(int argc, char** argv) 
{
    float x[3] = {-1,0,1};
    float y[3] = {7.059105, 24.998369, 14.365907};
    float p[3];
    
    polyf_fit(x,y,3,p,3);
    float max1 = -0.5*p[1]*p[2];
    
    polyf_fit_lagrange(x,y,3,p);
    float max2 = -0.5*p[1]*p[2];
    
    printf("%f and %f are different.\n", max1, max2);
    
    return 0;
}`

fsheikh, 2019-10-15:
Just came across this, but I think both documentation and code are correct.  I quickly added a [unit-test](https://github.com/fsheikh/liquid-dsp/commit/92ddb5ee95e8a3a887fc5ef4ba54b406b2720703) to confirm this. I think the  equivalence is meant in terms of error. The generated polynomials by lagrange and least-square method, for your input, are a mirror image.

[image]

fsheikh, 2019-10-20:
Sorry for the confusion but there was indeed a bug in lagrange polynomial. After the fix both least-square and lagrange method produce the same polynomials for given range of inputs.

pfeatherstone, 2019-10-20:
Jolly good

jgaeddert, 2019-10-21:
Thanks for the update. I'm aware that I'm sometimes inconsistent with my own definitions 😬

fsheikh, 2019-10-21:
@jgaeddert when you have can you please review #178 with the proposed fix. I will gladly rework the unit-tests if they are not up to your taste.

jgaeddert, 2019-10-21:
Thanks, @fsheikh! I might adjust the test a little in the future, but it's definitely better now.

closed 2019-10-21 by jgaeddert