jgaeddert/liquid-dsp #168: msourcecf, won't handle multiple modem modulation schemes

ltorlay, 2019-05-09:
I am trying to create a multisource channel with the msourcecf application in order to test a project I am working on. I am having an issue where I get the error "error: msourcecf_create(), invalid modulation scheme." This seems odd because it shouldn't have to define a single modulation scheme if you were to add multiple modem sources with different modulation schemes. Also msource documentation and msourcecf_example don't define a modulation scheme at the beginning (i.e. "msourcecf gen = msourcecf_create();"). 

Seems like it can't support mulitiple modulation schemes. Please correct me if I'm wrong.

Thank you!

jgaeddert, 2019-05-10:
Hmm. Definitely should be able to generate multiple independent sources. They have their own properties so different modulation schemes shouldn't be an issue.

I'm reworking the msourcecf framework now and can maybe merge that back in soon. The API will change, though.

ltorlay, 2019-05-13:
Thank you for your response! It actually was an issue with an array. I’m
sorry for the false alarm.

Love the software!

Thanks again,

Lucas Torlay

closed 2019-05-13 by ltorlay

jgaeddert, 2019-05-13:
No worries! I'm just glad it was resolved. Again, I'm changing the interface to be more extensible to different signal types, so that will mean you'll need to modify your code once those changes get merged in.

ltorlay, 2019-05-13:
Sounds good. Thank you