jgaeddert/liquid-dsp #169: firpfbch leakage ?

pfeatherstone, 2019-05-14:
If you modify the example firpfbch_crcf_analysis_example.c to generate a tone (at normalised frequency 0.3) with 1024 frames, you notice leakage in the output channels. I have attached the modified example so you can run it.
You will notice that channels 2 and 3 have a spike at normalised frequency 0.4. This doesn't seem correct to me.

[firpfbch_crcf_analysis_example_tone.zip](https://github.com/jgaeddert/liquid-dsp/files/3176854/firpfbch_crcf_analysis_example_tone.zip)

pfeatherstone, 2019-05-14:
Leakage is probably the wrong term. I would expect the tone to be present in channel 2 only.

pfeatherstone, 2019-05-14:
Ok i've implemented a polyphase filterbank channelizer (analysis) from scratch using [http://cradpdf.drdc-rddc.gc.ca/PDFS/unc03/p518656.pdf](url) and I get the same thing, i.e. a tone in channels 2 and 3. So I guess this implies there isn't a liquid bug. But I don't understand why channel 3 has the tone. Maybe this should be a stackoverflow type question

jgaeddert, 2019-05-17:
After some discussion through email, I think @pfeatherstone and I determined that a longer filter is necessary to improve channel isolation over the default `firpfbch_crcf_analysis_example.c` example.

closed 2019-05-17 by jgaeddert